mriscoc/Ender3V2S1 #176: mesh leveling doesn't work well

Marvindaut, 2022-05-29:
The mesh leveling curiously don’t work for me.
I got 4 models of the Ender 3 S1, two with the F103 and two with the F401 board. On every single printer the same problem:
It seems like it’s working (z-axis moves), but not the shape of the bed.
When I turn the ABL off, manually perfect levelt, it works way better.
On the following Picture you can see the results in comparison, the lower sheet is with ABL turned on.
Hopefully someone can help me.
BTW I also got the problem on stock firmware.

Marvindaut, 2022-05-29:
[image]

DevlshOne, 2022-05-29:
That sounds like a bad sensor to me.  Turn it off and everything works better?

Marvindaut, 2022-05-29:
> That sounds like a bad sensor to me. Turn it off and everything works better?

Can't believe it's the sensor, it would be veeery unlucky if I got four bad sensors:D
Yes the samples on the picture printed with same settings, except the upper sample was printed without using the mesh.
Deviation is about ~0.003.

carnaue, 2022-05-29:
I had same but fixed it with the UBL capabilities in the firmware. 

Instructions here: 

https://www.printables.com/model/213992-perfect-1st-layers-ender-3-v2-ender-3s1-ender-3s1-/files

mriscoc, 2022-06-01:
> Same here (latest firmware from here, CR Touch w highspeed turned off), getting a weird stripe in my mesh which is also occuring in the prints if mesh is turned on

Check the eccentric wheels of your X axis.

RoP09, 2022-06-01:
And here is the same 9x9:

[image]

mriscoc, 2022-06-01:
You have at least two positions in your X axis where the carriage doesn't move smoothly. Check your eccentric wheels and the head screws.

RoP09, 2022-06-02:
Ok, will check that. One more question, when I read the mesh in Octoprint it looks much better than the one shown in the printer:
[image]

Why is there such a discrepancy? Only view differences?
`Send: M420 V
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4      5      6      7      8
Recv:  0 +0.009 +0.027 +0.132 +0.035 +0.027 +0.077 +0.025 -0.073 +0.047
Recv:  1 -0.003 +0.020 +0.077 +0.012 +0.005 +0.050 +0.015 -0.070 +0.029
Recv:  2 -0.015 +0.032 +0.115 +0.047 +0.025 +0.065 +0.055 -0.008 +0.040
Recv:  3 -0.020 +0.030 +0.090 +0.020 +0.032 +0.100 +0.037 -0.060 +0.040
Recv:  4 -0.045 +0.022 +0.060 +0.005 +0.005 +0.057 +0.030 -0.075 +0.010
Recv:  5 -0.003 +0.032 +0.115 +0.022 +0.025 +0.080 +0.035 -0.065 +0.052
Recv:  6 +0.017 +0.040 +0.100 +0.042 +0.022 +0.077 +0.029 -0.034 +0.024
Recv:  7 -0.015 +0.022 +0.081 +0.012 +0.020 +0.077 +0.052 -0.075 +0.037
Recv:  8 +0.015 +0.035 +0.097 +0.032 +0.020 +0.070 +0.057 -0.048 +0.032
Recv: 
Recv: echo:Bed Leveling OFF
Recv: echo:Fade Height 10.00
Recv: ok P15 B3`

RoP09, 2022-06-02:
Did again 2 test  prints - a single 0.2mm layer and laying them out on a light photo plane so it shines through to make the densitiy of the material more visible , first one with mesh leveling (9x9) enabled, 2nd one without. 
Why is the one without actually more uniform, would not expect that at all?
[image]

Marvindaut, 2022-06-02:
I suspect the following in the meantime:
Due to the fact that, not the entire bed measured during automatic leveling (40mm missing at the back), a reasonable mesh cannot be created. Creality should have made the profile for the Y-axis 40mm longer, so that not only the entire bed can be printed, but also measured.
With the automatic leveling, it is then somehow converted, since the missing areas can only be estimated. This would also explain why a perfectly, manually leveled bed is "corrected" again during an automatic mesh built. Am I correct with this?

mriscoc, 2022-06-02:
> Ok, will check that. One more question, when I read the mesh in Octoprint it looks much better than the one shown in the printer (green = good?)

@RoP09, I don't see the discrepancy, indeed the Octoprint mesh visualizer confirms the printer mesh viewer data, about the numbers, don't forget that, in the matrix from the plugin, the position 0,0 is in the upper left corner.

RoP09, 2022-06-02:
> @RoP09, I don't see the discrepancy, indeed the Octoprint mesh visualizer confirms the printer mesh viewer data, about the numbers, don't forget that, in the matrix from the plugin, the position 0,0 [...]

No I mean according to Octorpint everything is "good" (green), still the ABL is not able to successfully level the print. How much better has the bed to be to make it level succssfully? 
Or in other words, what is the reason for having ABL if it cannot deal with this level of bed level shifts?

mriscoc, 2022-06-03:
Check this https://www.printables.com/es/model/213992-perfect-1st-layers-ender-3-v2-ender-3s1-ender-3s1-****

RoP09, 2022-06-03:
> Check this https://www.printables.com/es/model/213992-perfect-1st-layers-ender-3-v2-ender-3s1-ender-3s1-****

I read that before, but this would basically mean i can take the CR Touch and throw it out the window.

Victordiazbis, 2022-06-03:
Yes and no. The problem comes from the kinematics of the printer, not from the CR touch. As the CR touch is quite far from the nozzle, any imperfection produces very bad results in the mesh. Check every element of your X axis, mainly the wheels.

Victordiazbis, 2022-06-03:
> > Yes and no. The problem comes from the kinematics of the printer, not from the CR touch. As the CR touch is quite far from the nozzle, any imperfection produces very bad results in the mesh. Check every element of your X axis, mainly the wheels.
> 
> Even everything is perfekt adjusted on the printer, the problem of the "unknown 40mm", stays. I just ordered a 2040 v-slot profile to extend the range of the y axis by 40mm. According to my understanding, that should solve the main problem. I will report my experience, when installed.
> 
> By the way another question, why can't I change the bedsize? I tried to change to 230x230, but on x axis the firmware seems to fail and on y axis, its limited to 220. Is there any reason for that, and is it possible to fix/ unlock it?

I would ask to Miguel for a possible improvement. The chance extend the auto mesh to the unmeasurable area by adding points defined manually.

Victordiazbis, 2022-06-03:
If I have time I would like to design a bracket to relocate the CR touch closer to the nozzle. I think everything would improve if x & y offsets weren't so big

RoP09, 2022-06-03:
> If I have time I would like to design a bracket to relocate the CR touch closer to the nozzle. I think everything would improve if x & y offsets weren't so big

Indeed i have a custom mount made for the CR Touch which is pretty far away ( -50/-30)

mriscoc, 2022-06-03:
> By the way another question, why can't I change the bedsize? I tried to change to 230x230, but on x axis the firmware seems to fail and on y axis, its limited to 220. Is there any reason for that, and is it possible to fix/ unlock it?

Did you change the X max position? The bed can't be bigger than the maximum allowed axis size.

mriscoc, 2022-06-08:
Check this: https://github.com/mriscoc/Ender3V2S1/issues/154

closed 2022-06-08 by mriscoc

Marvindaut, 2022-06-09:
Ok I now upgraded my printer by a profile for the y axis.
 
[image]

I tested some configurations and now the sensor is measuring exactly centered and on the same Points, where the nozzle prints.
On this picture you can see a printed grid with 9x9 squares (220mmx220mm). 

[image]

I changed to the following settings:
Physical settings:
        X bed size: 230
        Y bed size: 230
        X min position: -5
        Y min position: -3
        X max position: 252
        Y max position: 270
Mesh inset:
        Mesh X minimum: 10
	Mesh Y minimum: 10
	Mesh X maximum: 220
	Mesh Y maximum: 220
So far so good, but I still got problems on the first Layer.
Here two pictures of the mesh and a first layer print:

[image]
[image]

As you can see, the nozzle was too close to the bed at the front, and further back it was partly too far away. This problem is still on my other S1 – front to close, back to far away.
Do I have to change anything else in the firmware?
Is there any offset that maybe shifts the mesh, so that it is not perfect placed on the bed?
Thank you for your effort!

RoP09, 2022-06-09:
Same here in the E3V2. Exchanged the eccentric wheels and created a new mount for the CR Touch now sitting at x=0 to the nozzle.

Still the same stripes as shown in the pictures above.

Victordiazbis, 2022-06-10:
> Same here in the E3V2. Exchanged the eccentric wheels and created a new mount for the CR Touch now sitting at x=0 to the nozzle.
> 
> Still the same stripes as shown in the pictures above.

Yesterday I replaced the wheels and the CR-Touch. Mesh looks good with no stripes, however  nozzle is slightly closer to the bed in the right side of the bed. I have to tilt the bed after mesh creation to get good results. Maybe my x axis profile is slightly twisted. Wheels and auto bed leveling are not good combination. Maybe with linear rails