mriscoc/Ender3V2S1 #178: Just an question

natewils, 2022-05-31:
### Description

### Steps to Reproduce

1. [Test Print after install]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [ Not to bottom out so to speak at both ends of X axis]

**Actual behavior:** [ Homes fine but during probing and purge line the carriage hits hard stop on the right side and bounds up to endstop on the left side during purge line ]

I installed [Ender3V2-422-BLTUBL-20220527.bin] and coming from Jyers i am figuring it out but when probing it hits the x axis hardstop/end to the right and on the purge line side it hits endstop and trys to keep going , is there something i need to change and where thanks in advance .

mriscoc, 2022-05-31:
If you have a custom fan duct, you must change the XY Max positions according to your real printer physical limits.

natewils, 2022-05-31:
I will be looking to this , it was never a problem and maybe I was just lucky .   Thanks again

mriscoc, 2022-06-01:
Other firmwares have conservative limits, the default settings of this firmware maximize the work area of a stock printer. If the printer is non-stock you must reconfigure the physical limits: https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides

closed 2022-06-08 by mriscoc