mriscoc/Ender3V2S1 #184: Modifying Probe Offset Causes Tramming Wizard to Not Work Correctly

Rodman101, 2022-06-02:
Description:
After modifying the probe offset, the "Tramming Wizard" feature will not probe all four corners of the bed.

Steps to reproduce the behavior:
1. Change Probe Offset Values
   X Offset = -45
   Y Offset = -8
3. Store Settings
4. Use Tramming Wizard

Expected Behavior:
Probe will test all four corners of the bed. Front Left, then Front Right, then Rear Right, then Rear Left.

Actual Behavior:
Front Left will be probed. The display will blink twice, then proceed to probe the Rear Left. It shows the Right corners as being at 0.0

Firmware version and release date:
Ender3V2-422-BLTUBL-20220527

mriscoc, 2022-06-02:
Please check your mesh insets in Advanced / Mesh Leveling / Mesh Insets
You can use Maximize Mesh and then Center Mesh to fix it automatically.

closed 2022-06-02 by mriscoc

RoP09, 2022-06-09:
> Please check your mesh insets in Advanced / Mesh Leveling / Mesh Insets You can use Maximize Mesh and then Center Mesh to fix it automatically.

I have the same issue, the Mesh insets are either calculated wrong, or the printer applys them in the wrong way, as the mesh is neither centered correctly, nor are the borders aligned  with the same offsets (e.g. left and right are different).

mriscoc, 2022-06-10:
Check your physical settings.