mriscoc/Ender3V2S1 #186: Printer reboot on accessing print menu and creat jpeg thumbnail script not working cura 5

Moobien, 2022-06-04:
**Describe the bug**
A clear and concise description of what the bug is.

Since updating to firmware BLT 20220527 my printer will reboot every time I try to access the print menu for 3-5 minutes after inserting my micro sd card.

**To Reproduce**
Steps to reproduce the behavior:
1. insert micro sd card
2. wait a few seconds
3. try to open print menu

**Expected behavior**
A clear and concise description of what you expected to happen.

previously i could go to my print menu seconds after inserting my micro sd card.

Create JPEG Thumbnail script issue.

installed the script as per instructions for cura 5, however when you go to extenssions - post processing etc to enable the script it does not show in the list to be selected.

mriscoc, 2022-06-05:
Format or change your memory card. 
Update the post processing script.

Moobien, 2022-06-05:
Have reformatted the card numeous times,

The updating the script worked! thank you for that,