mriscoc/Ender3V2S1 #193: Occasionaly reporting incorrect high temps to octoprint

melvinmicallef, 2022-06-10:
**Describe the bug**
I am currently using octoprint with my ender 3v2 4.2.2 board running 20220527. Occasionally, whilst printing, the printer reports a much higher temperature(hotend) than what the printer is actually at. This happened twice to me since i updated to the latest firmware. 

From the following logs, you can see that the temperature reporting is normal, apart from the last line, where in 2 seconds, it is as if the temperature went up to from 205 to 255. 

2022-06-11 01:09:49,910 - Recv:  T:204.97 /205.00 B:59.97 /60.00 @:80 B@:36
2022-06-11 01:09:51,911 - Recv:  T:205.11 /205.00 B:60.02 /60.00 @:77 B@:24
2022-06-11 01:09:53,925 - Recv:  T:205.08 /205.00 B:60.04 /60.00 @:78 B@:21
2022-06-11 01:09:55,916 - Recv:  T:204.95 /205.00 B:60.03 /60.00 @:81 B@:23
2022-06-11 01:09:57,923 - Recv:  T:204.94 /205.00 B:60.01 /60.00 @:81 B@:29
2022-06-11 01:09:59,915 - Recv:  T:204.86 /205.00 B:60.01 /60.00 @:82 B@:30
2022-06-11 01:10:01,916 - Recv:  T:204.94 /205.00 B:60.02 /60.00 @:80 B@:26
2022-06-11 01:10:03,915 - Recv:  T:204.95 /205.00 B:60.03 /60.00 @:80 B@:22
2022-06-11 01:10:05,915 - Recv:  T:204.76 /205.00 B:60.01 /60.00 @:84 B@:26
2022-06-11 01:10:07,915 - Recv:  T:204.97 /205.00 B:60.03 /60.00 @:80 B@:23
2022-06-11 01:10:09,915 - Recv:  T:204.85 /205.00 B:60.01 /60.00 @:82 B@:27
2022-06-11 01:10:11,915 - Recv:  T:204.90 /205.00 B:60.03 /60.00 @:81 B@:22
2022-06-11 01:10:13,915 - Recv:  T:204.99 /205.00 B:60.03 /60.00 @:80 B@:22
2022-06-11 01:10:15,915 - Recv:  T:204.95 /205.00 B:59.98 /60.00 @:81 B@:32
2022-06-11 01:10:17,915 - Recv:  T:204.84 /205.00 B:60.00 /60.00 @:83 B@:28
2022-06-11 01:10:19,915 - Recv:  T:204.87 /205.00 B:60.02 /60.00 @:83 B@:22
2022-06-11 01:10:21,916 - Recv:  T:204.82 /205.00 B:60.00 /60.00 @:84 B@:27
2022-06-11 01:10:23,915 - Recv:  T:205.06 /205.00 B:59.99 /60.00 @:79 B@:28
2022-06-11 01:10:25,917 - Recv:  T:205.19 /205.00 B:60.00 /60.00 @:77 B@:27
2022-06-11 01:10:27,915 - Recv:  T:205.24 /205.00 B:60.00 /60.00 @:76 B@:26
2022-06-11 01:10:29,915 - Recv:  T:205.19 /205.00 B:59.95 /60.00 @:78 B@:38
2022-06-11 01:10:31,915 - Recv:  T:205.39 /205.00 B:59.98 /60.00 @:73 B@:32
2022-06-11 01:10:33,915 - Recv:  T:205.32 /205.00 B:60.02 /60.00 @:75 B@:23
2022-06-11 01:10:35,915 - Recv:  T:205.25 /205.00 B:60.01 /60.00 @:75 B@:27
2022-06-11 01:10:37,915 - Recv:  T:204.95 /205.00 B:60.03 /60.00 @:81 B@:23
2022-06-11 01:10:39,915 - Recv:  T:205.09 /205.00 B:60.02 /60.00 @:77 B@:24
2022-06-11 01:10:41,915 - Recv:  T:204.86 /205.00 B:60.04 /60.00 @:82 B@:20
2022-06-11 01:10:43,914 - Recv:  T:204.75 /205.00 B:60.01 /60.00 @:84 B@:26
2022-06-11 01:10:45,914 - Recv:  T:204.89 /205.00 B:60.00 /60.00 @:80 B@:29
2022-06-11 01:10:47,915 - Recv:  T:204.93 /205.00 B:59.98 /60.00 @:79 B@:33
2022-06-11 01:10:49,914 - Recv:  T:204.90 /205.00 B:59.96 /60.00 @:81 B@:36
2022-06-11 01:10:51,915 - Recv:  T:205.03 /205.00 B:60.01 /60.00 @:78 B@:25
2022-06-11 01:10:53,914 - Recv:  T:204.99 /205.00 B:59.98 /60.00 @:79 B@:33
2022-06-11 01:10:55,914 - Recv:  T:205.04 /205.00 B:60.03 /60.00 @:78 B@:21
2022-06-11 01:10:57,914 - Recv:  T:205.09 /205.00 B:60.00 /60.00 @:78 B@:28
2022-06-11 01:10:59,914 - Recv:  T:205.08 /205.00 B:60.00 /60.00 @:78 B@:28
2022-06-11 01:11:01,914 - Recv:  T:205.08 /205.00 B:59.99 /60.00 @:78 B@:29
2022-06-11 01:11:03,914 - Recv:  T:205.07 /205.00 B:59.99 /60.00 @:78 B@:28
2022-06-11 01:11:05,914 - Recv:  T:204.99 /205.00 B:59.99 /60.00 @:79 B@:28
2022-06-11 01:11:07,914 - Recv:  T:205.00 /205.00 B:60.00 /60.00 @:79 B@:25
2022-06-11 01:11:09,914 - Recv:  T:204.93 /205.00 B:60.01 /60.00 @:80 B@:23
2022-06-11 01:11:11,914 - Recv:  T:204.90 /205.00 B:59.98 /60.00 @:80 B@:31
2022-06-11 01:11:13,916 - Recv:  T:204.84 /205.00 B:59.98 /60.00 @:82 B@:31
2022-06-11 01:11:15,914 - Recv:  T:204.91 /205.00 B:59.98 /60.00 @:80 B@:30
2022-06-11 01:11:17,920 - Recv:  T:205.00 /205.00 B:59.96 /60.00 @:79 B@:36
2022-06-11 01:11:19,914 - Recv:  T:205.03 /205.00 B:59.98 /60.00 @:78 B@:31
2022-06-11 01:11:21,914 - Recv:  T:205.18 /205.00 B:59.99 /60.00 @:76 B@:30
2022-06-11 01:11:23,914 - Recv:  T:205.19 /205.00 B:59.99 /60.00 @:76 B@:32
2022-06-11 01:11:25,914 - Recv:  T:205.35 /205.00 B:59.98 /60.00 @:73 B@:34
2022-06-11 01:11:27,913 - Recv:  T:205.00 /205.00 B:59.98 /60.00 @:80 B@:35
2022-06-11 01:11:29,915 - Recv:  T:205.21 /205.00 B:60.05 /60.00 @:76 B@:20
2022-06-11 01:11:31,913 - Recv:  T:205.08 /205.00 B:60.03 /60.00 @:78 B@:24
2022-06-11 01:11:33,913 - Recv:  T:205.00 /205.00 B:59.98 /60.00 @:79 B@:34
2022-06-11 01:11:35,913 - Recv:  T:205.06 /205.00 B:60.03 /60.00 @:77 B@:22
2022-06-11 01:11:37,913 - Recv:  T:204.90 /205.00 B:60.01 /60.00 @:80 B@:26
2022-06-11 01:11:39,917 - Recv:  T:204.85 /205.00 B:60.03 /60.00 @:81 B@:21
2022-06-11 01:11:41,913 - Recv:  T:204.88 /205.00 B:60.01 /60.00 @:80 B@:24
2022-06-11 01:11:43,913 - Recv:  T:204.77 /205.00 B:59.98 /60.00 @:83 B@:31
2022-06-11 01:11:45,914 - Recv:  T:204.81 /205.00 B:60.01 /60.00 @:82 B@:24
2022-06-11 01:11:47,913 - Recv:  T:204.94 /205.00 B:59.96 /60.00 @:79 B@:36
2022-06-11 01:11:49,913 - Recv:  T:204.90 /205.00 B:59.98 /60.00 @:81 B@:31
2022-06-11 01:11:51,913 - Recv:  T:204.92 /205.00 B:60.01 /60.00 @:81 B@:26
2022-06-11 01:11:53,913 - Recv:  T:205.18 /205.00 B:59.99 /60.00 @:76 B@:31
2022-06-11 01:11:55,913 - Recv:  T:205.26 /205.00 B:60.01 /60.00 @:75 B@:26
2022-06-11 01:11:57,917 - Recv:  T:205.28 /205.00 B:60.00 /60.00 @:75 B@:29
[... 108 lines cut ...]
2022-06-11 01:15:35,923 - Recv:  T:205.31 /205.00 B:60.01 /60.00 @:74 B@:27
2022-06-11 01:15:37,925 - Recv:  T:205.18 /205.00 B:60.03 /60.00 @:77 B@:21
2022-06-11 01:15:39,919 - Recv:  T:205.12 /205.00 B:60.00 /60.00 @:78 B@:27
2022-06-11 01:15:41,919 - Recv:  T:205.07 /205.00 B:60.02 /60.00 @:78 B@:22
2022-06-11 01:15:43,920 - Recv:  T:204.96 /205.00 B:60.01 /60.00 @:80 B@:25
2022-06-11 01:15:45,919 - Recv:  T:204.95 /205.00 B:59.99 /60.00 @:80 B@:29
2022-06-11 01:15:47,919 - Recv:  T:204.91 /205.00 B:60.03 /60.00 @:80 B@:19
2022-06-11 01:15:49,919 - Recv:  T:204.93 /205.00 B:60.00 /60.00 @:80 B@:26
2022-06-11 01:15:51,919 - Recv:  T:255.00 /205.00 B:59.99 /60.00 @:78 B@:29 - Octoprint thermal runaway triggered due to high temp
 

**To Reproduce**
Not sure but i printed two different files and the issue happened randomly. I am thinking this is a firmware issue and not a faulty thermistor since the numbers seem to have very similar digits (205 and 255) and the value is a whole number. My config is also set to have a maximum temperature of 235 and an overshoot of 15, so my printer can only handle a temp of 250 before shutting down. 

**Expected behavior**
Temperature to be reported consistently. Kindly check if there is an issue with temperature reporting, or let me know how i can provide you with further info.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version (please complete the following information):**
20220527

**Additional context**
Add any other context about the problem here.

mriscoc, 2022-06-11:
I use the 20220527 4.2.2 BLT UBL version with the same Tasmota plugin, I don't have that bug in my setup.

melvinmicallef, 2022-06-11:
Any idea what this might be please? It is highly unlikely that this is a thermistor issue. I have performed PID autotune as well but issue persists.
[image]
[image]

closed 2022-07-05 by mriscoc