mriscoc/Ender3V2S1 #194: Advanced Pause not working the same as previous prof firmware

carnaue, 2022-06-11:
**Describe the bug**
In cura I use the pause at layer feature to insert magnets into prints.  I have previously been doing this fine with the mriscoc firmware.  The pause at height Cura script sends the following information to the printer.

```
;current layer: 8
M117 Insert Magnet
M0 ; Do the actual pause
;LAYER:8
```

This then pauses the print and moves the print head away from the bed so you can access the print.  Now the exact same gcode pauses the print but does not move the print head away from the print as it used to.

**To Reproduce**
Steps to reproduce the behavior:
1. Start a print
2. Send an M0 command to the printer during a print

**Expected behavior**
Print should pause and print head should move to a safe location so you can access the print.  On resume it should go back to the print and continue from where it was paused.

**Screenshots**
[image]

**Version (please complete the following information):**
Ender3 V2/S1 Professional Firmware 20220527

**Additional context**
Worked fine on a previous version of the firmware.

mriscoc, 2022-06-13:
For Advanced Pause use the [M25](https://marlinfw.org/docs/gcode/M025.html) or [M125](https://marlinfw.org/docs/gcode/M125.html) G-code, [M0](https://marlinfw.org/docs/gcode/M000-M001.html) is a simple pause.

carnaue, 2022-06-14:
Updated octoprint scripts for before pause and after pause from the details on the wiki https://github.com/mriscoc/Ender3V2S1/wiki/Octoprint.  Using M25 and it now works as intended.  Thanks Miguel.

closed 2022-06-14 by carnaue