mriscoc/Ender3V2S1 #195: Z-offset affects bed mesh creation

abra5umente, 2022-06-11:
**Describe the bug**
When levelling with a z-offset of anything other than 0, it affects the way the mesh is produced. Consistently reads one side MUCH higher than the other, seems to flip around from side to side too.

**To Reproduce**
Steps to reproduce the behavior:
1. Perform a manual tramming using a feeler gauge or piece of paper.
2. Set a Z-offset in the menu (either by babystepping or by using the wizard)
3. Perform an auto level procedure (I leave the bed off, but doesn't seem to make a difference)
4. Note how one side will be particularly higher than the other, or there will be random valleys and hills in the mesh that you know aren't there.
5. Clear the z-offset back to 0
6. Restart the printer
7. Home the printer
8. Perform a bed level procedure again
9. Note how the mesh is more "true to life" and the results are much different.

**Expected behavior**
The z-offset should not affect the measurements of the bed.

**Screenshots**
[image]

**Version (please complete the following information):**
Latest firmware, 2.1.0.1, build date May 27th 2022. Ender 3 S1 with F4 chip.

**Additional context**
Have tried using BLTouch with same results. Tried with heating and not heating the bed, have verified that there is no power backfeed from Raspberry Pi 4 to printer (using electrical tape over the power pins).

nilsafendi, 2022-06-11:
I have been having the hardest time trying to get a level bed with my S1. The bottom left quartile was always smushed and the upper right quartile was the opposite. There was also a random vertical dip that spanned from the middle of the bottom and upper left quartiles. 

I'm on vacation without access to my printer but I'm looking forward to trying this.

abra5umente, 2022-06-11:
Here is that same bed, same printer, same room temperature, but with a z-offset of -2.35 set while probing.

[image]

And here it is again after probing with z-offset set to 0 and rebooting the printer.

[image]

elkinator1, 2022-06-11:
> [image: My glass bed with a smooth, flat surface, same as read on my other
> printer with a BLTouch. I have also verified the CRTouch on that printer
> which performs the same as the BLTouch]

I recently got a CR Touch for my Ender 3 v2. Everything went down hill from
there. The stock firmware the CR Touch recommends moves my x-axis in the
wrong direction. I switched to this firmware by someone's recommendation
and it has been constant failed prints. After all sorts of troubleshooting
I went back to manual leveling with a guage. Problem solved.

Prior to going back to manua leveling, I would tram with a heated bed and
hotend. I would make adjustments until all 4 corners were exactly the same.
Then do my Z offset, then ABL. Everything seemed perfect on the software
end minus the center of my bed being lower. With all the youtubers and
comments out there saying the beds are wrapped and solved by the ABL. I
looked passed the warp bed until I was finally ready to print and it just
kept failing.

I kept finding (according to the CRT) that the center was always lower no
matter how much I tightened up the 4 corners. At the time, I had 25mm
yellow springs that I maxed out the tension on to try and at least bring
the 4 corners below center. None of this mattered of course but it was part
of my troubleshooting journey. From there I got blue 30-35mm blue springs
and maxed them out. I made sure my glass was not lifting from the bed at
all and the results were the same, the center of my bed was some how still
lower, I couldn't believe it.

From there I gave up and set my printer to its z-home position before
disabling steppers. I moved the x and y axis and found my bed was much
lower on the right side (which made sense with the way my printer was
failing). I decided to just try manually leveling again and then ran the
bed traming and the ABL/UBL with 8 touch points.

The results of both showed a fairly level bed (even the center) up until
the last row on right side, which according to the CRT it was way to high.

The end, I either have a defective CRT (which I doubt), I'm doing something
wrong, or there is a bug in the firmware.

abra5umente, 2022-06-11:
> I recently got a CR Touch for my Ender 3 v2. Everything went down hill from there. The stock firmware the CR Touch recommends moves my x-axis in the wrong direction. I switched to this firmware by [...]

I don't think the CRTouch is bad hardware - from my tests it's actually more accurate than the BLTouch. I don't know what is going on with the firmware to make bed levelling screw up so bad though - I've used ABL many, many times and on my Ender 3 Pro (with the same glass bed!) I could just plonk the bed on, make the corners relatively level, run G29 before my print and put M420 S1 in the start up code and it would fire away and I'd have perfect first layers, every single time.

The only differences I can see in the firmwares (I compiled my own firmware on the Ender 3 Pro and downloaded the repo to see what this one is doing) is that they have the multiple probes enabled, which I turned off. I don't see how exactly hardware could make the ABL so dodgy, but it could be that it's some strange issue with the F4 chip, or something. 
I'm going to compile firmware with multiple probes disabled and see if that makes a difference, you typically only need that enabled when your probe is crappy and you need higher accuracy.

abra5umente, 2022-06-11:
Alright, so I've compiled my own firmware with multiple probing disabled, and the result is much better.
1. It is literally 3 times faster, which is nice.
2. Z-offset was set to -2.35 here and the result is basically the same as when it was 0 with stock settings.
[image]

abra5umente, 2022-06-11:
New firmware is here, if anyone wants to test it out.
https://github.com/abra5umente/Ender3S1F1-4/releases/tag/20220527_probefix

webmogul1, 2022-06-13:
If by disabling multiple probing solved the problem, it could be that the probe is not being consistent.
Have you tried testing the probe? (Advanced Menu) (M48 Probe Test) 
It will show the probe deviation.

mriscoc, 2022-06-13:
M48 should be done at the same position where the errors are occurring, the deviation is not the real accuracy because if one of the measurements is out of the other 10, it does not affect too much to deviation. More interesting is the **Range value**.

abra5umente, 2022-06-13:
It occurs in random positions every level - sometimes on the right side, sometimes on the left side. I've been through the configuration.h file and I can't see anywhere that would skew any results. A few people have said that simply reducing it to one fixes it - so I'm assuming that it's an electromagnetic interference issue that just compounds with each press. 
I did try with the bed heated and not heated, and it made no real difference. 
Do you have an S1 to test with?

mriscoc, 2022-06-13:
I don't have a S1, that's why that error is difficult to reproduce and fix for me. But I have reports from other S1 users that don't have errors, maybe it could be a problem with the ground.

abra5umente, 2022-06-13:
I don't think it's a problem with ground, but I'll have a play tomorrow. I've had another printer with a BLTouch plugged into that same port for years and it's been fine - but I'll see what I can come up with.

trulow, 2022-06-14:
I also am noticing an issue with the CR-Touch. I find that probing isn't accurate and the results are jumping around by  .02 +/- deviation. You can really notice this when using the tramming wizard as the results are never consistent especially after a printer restart.

mriscoc, 2022-07-05:
> I also am noticing an issue with the CR-Touch. I find that probing isn't accurate and the results are jumping around by .02 +/- deviation. You can really notice this when using the tramming wizard as the results are never consistent especially after a printer restart.

It could be not only the CR-touch, the mechanical accuracy of the printer is 0.1 mm then if the printer moves is possible to have large deviations.

albertoabrao, 2022-07-10:
> New firmware is here, if anyone wants to test it out. https://github.com/abra5umente/Ender3S1F1-4/releases/tag/20220527_probefix

Thank you for sharing the result of your work!

@mriscoc, my printer is an Ender 3 S1 (F4). I can report having the exact same issue reported by  @abra5umente , and his version of the firmware solves the problem.

I assume it has something to do with the lack of memory, some kind of overflow happening when you try to store too many values? Not very familiar with Marlin, just a wild guess.

The delta between corners and center was 0.1, now it's 0.02. That's without touching a single screw, just new firmware. All else is the same (temperature etc).

It being faster is certainly nice too. The CR Touch is consistent enough otherwise, as reported by M48 Test - which incidentally also gives me a lower margin of error now.

mriscoc, 2022-07-11:
The next version will allow to edit the multiple probe value.

Moobien, 2022-07-17:
I have had something similar.  I will Tram my bed then set my z offset using a feeler gauge.  then after G29 has run on a print the nozzle seems to be a lot higher than what it was set at in the z offset, often to the effect of having to lower it up to 0.12mm (and I use a .1mm feeler.

nmanclank, 2022-07-18:
> I have had something similar. I will Tram my bed then set my z offset using a feeler gauge. then after G29 has run on a print the nozzle seems to be a lot higher than what it was set at in the z offset, often to the effect of having to lower it up to 0.12mm (and I use a .1mm feeler but also notice a range of -0.12 to even 0 on the mesh

I'm having this exact same issue.

GRRK, 2022-07-19:
Is it possible that X axis is inverted in auto leveling? I have 3S1 F4. My auto leveling seems to apply wrong offset the way bad is tilted in X.
In last test I ran 4x4 auto leveling and then manually mirrored the valued on X.... finally I can use whole bed with this approach.

hayden-t, 2022-07-31:
to test, adjust bed screws to extreme opposites, to create maximum slant, run abl, and then comand move the head around to see if it follows the slant correctly

arthurkok2, 2022-08-01:
on this line in creality's S1 firmware, they have BLTOUCH_DELAY configured. This may cause some differences?

https://github.com/CrealityOfficial/Ender-3S1/blob/main/Marlin/Configuration_adv.h#L771

Moobien, 2022-08-01:
> on this line in creality's S1 firmware, they have BLTOUCH_DELAY configured. This may cause some differences?
> 
> https://github.com/CrealityOfficial/Ender-3S1/blob/main/Marlin/Configuration_adv.h#L771

Could that be the difference between HS mode enabled/disabled in this firmware?

arthur-ceridian, 2022-08-01:
> > on this line in creality's S1 firmware, they have BLTOUCH_DELAY configured. This may cause some differences?
> > https://github.com/CrealityOfficial/Ender-3S1/blob/main/Marlin/Configuration_adv.h#L771
> 
> Could that be the difference between HS mode enabled/disabled in this firmware?

They seem to have HS mode enabled too https://github.com/CrealityOfficial/Ender-3S1/blob/main/Marlin/Configuration_adv.h#L816

hayden-t, 2022-08-01:
HS mode works on my S1 pro, and without it probe does not work. I can do 9x9 or any size with HS, and it matches my bed contour.

mriscoc, 2022-08-01:
> on this line in creality's S1 firmware, they have BLTOUCH_DELAY configured. This may cause some differences?
> 
> https://github.com/CrealityOfficial/Ender-3S1/blob/main/Marlin/Configuration_adv.h#L771

If you don't define a delay there is a default value of 500ms
https://github.com/CrealityOfficial/Ender-3S1/blob/e46e1a85011fb91cd8f8c1cc9c5a1da6887f045f/Marlin/src/inc/Conditionals_post.h#L2926

mriscoc, 2022-08-01:
> They seem to have HS mode enabled too https://github.com/CrealityOfficial/Ender-3S1/blob/main/Marlin/Configuration_adv.h#L816

They enable HS because their mesh is only 4x4

harley44howard, 2022-08-02:
I'm having similar issues with ender 3v2, sprite pro, dual z. Having a hard time leveling,  can get zeros on all 4 corners  using bed tramming. When do a mesh, no matter the size, I end up getting obscure high and low spots. Set z offset and print a large flat square to live adjust z offset while in printing the skirt I can see how one side of my bed is higher than the other.. I have squish on one side and to far away from the bed on the other.  My only work around was doing what the original post was about and setting zoffset to zero them doing bed tramming,  auto mesh,  edit mesh manually then set z offset and live adjust.. it Is nowhere near perfect but I'm atleast printing now.  I also level the bed turning the knobs what printing because I noticed still one side to high in comparison to the other.. Basically ignored my mesh and adjusted z offset and leveling knobs while watching the first layer of a print... I also have linear y axis and 3 point leveling.. adding the 4 point back didn't change anything.. it is like the probe becomes inaccurate or just doesn't actually set the mesh during the print anymore.. yes I have the proper gcode set, and it does load the mesh before the print. I also tried to tilt the mesh before, and without tilting.

hayden-t, 2022-08-02:
"They enable HS because their mesh is only 4x4"
HS mode completes upto 9x9 ...

@harley44howard - the problem is the magnetic medium layer is not even (put a ruler edge across it), and even though the probe detects this  it does not seem to translate into enough compensation (im still investigating why) . as a manual temp solution just put some foil under between the build plate and mag layer in the low spots

harley44howard, 2022-08-02:
> "They enable HS because their mesh is only 4x4" HS mode completes upto 9x9 ...
>
> @harley44howard - the problem is the magnetic medium layer is not even (put a ruler edge across it), and even though [...]

@hayden-t I'll take a gander at that. It was quite severe though and if I remember correctly it's the same bed and magnetic pad from the firmware before this without any issues. Also tried different pei sheets (2 new ones) I have 2 new magnetic sheets as well I guess I could try a new one.

albertoabrao, 2022-08-02:
I noticed that the original firmware for the S1 uses the following feedrates:

Recv: echo:; Maximum feedrates (units/s):
Recv: echo:  M203 X300.00 Y300.00 Z5.00 E25.00

Professional Firmware uses Z20 or so.

I've been getting more consistent results by using the same values as mentioned above, on the Professional Firmware, of course - Creality's is just terrible. Tried raising Z to 10, and it was significantly worse already.

albertoabrao, 2022-08-05:
> that is your max z speed, do you think the leveling runs at max z ? im not so sure

I did not think much, I just changed it to see what would happen.

The end result was that I got a slower probe movement, both up and down the axis,  than I was able to by reducing the probe speed - there, it would reduce the downwards speed, but not so much the reverse.

It also resulted in more consistent probing results, so I decided to keep it. And share, in case it helps someone else.

Now, as to why this is the case - or not, in case I am wrong - mriscoc or someone else more knowledgeable than I am may be willing to elucidate.

jordanpenard, 2022-08-19:
I'm glad I found that thread, because I was about to loose my mind with the auto bed leveling on my S1 (F1) . I'm seeing similar issues than what @abra5umente reports, even though I didn't try to run the ABL with zoffset at 0.
Where I am so far is that I ran multiple 9x9 ABL, plotted the results in octoprint bed level display, and tried to find the tallest and lowest reported points on my machine while the level adjustment is enabled, and everything is out of place, the nozzle distance to the bed changes massively between those points, by around 0.2/0.3 mm, it even caused a few holes in my bed surface.
My next step is to do a manual bed leveling and recorde le values to compare with the ABL values.
I will try the FW that @abra5umente posted earlier too.
What I'm still unsure of :  
- is the ABL amplifing the values measured and ends up over compensating? 
- are X and Y swapped around? 

I shall come back with my findings once I get the time to do more testing.

hayden-t, 2022-08-20:
please look at my 2 latest videos regarding abl investigations https://www.youtube.com/user/hthring

mriscoc, 2022-08-20:
> please look at my 2 latest videos regarding abl investigations https://www.youtube.com/user/hthring

You need to consider the weight of the head and the impact on the deformation of the X axis.

hayden-t, 2022-08-20:
i think there are several reason the plane of travel can become skewed

JFingerle, 2022-09-23:
Is this still an issue an F4 boards? Thank you!

DanMan32, 2022-10-31:
Could the problem everyone (or most) are seeing is due to too much slop in the Z axis, especially on the right side?
The z axis on the Enders is supposed to be 0.04mm.   When manually tramning corners even before using this firmware, I noticed that there was a z-offset deviation when moving the Z up vs down.  I've also noticed that when printing 5 squares bed level/z-offset test, the left side may be perfect but right side not as good.
Most Z movement during a print is in the up direction, but mesh building using ABL is read in the down direction, so right there  there is a deviation in the mesh offset vs what actually exists during print.

Then again the mesh display is based completely on ABL readings, but if there's Z axis slop, that would explain the deviations between readings.

mhaas, 2022-11-06:
Just to add one data point to the discussion. I am trying to figure out some bed adhesion issues on my S1 and suspected that the ABL is an issue for me. I tried the "disable multiple probing" fix described here on latest master with the S1 UBL config.

Results with stock professional:
[image]

Results without multiple probing:
[image]

The actual numeric values are different, but the overall shapes seems to have been retained.

Will go back to stock professional for now; my first layer is still not sticking and I will try other means of fixing that.

DanMan32, 2022-11-06:
You do seem to have a lot of variation, but as you said, the pattern is close.

Not sticking to the bed typically would be due to z-offset with too much of a gap, dirty bed (clean with soap/water and/or IPA) or bed/nozzle temp too low.

Remember, the probe is measuring z height of the bed and the probe, not the nozzle.  The z-offset sets the difference between the prone '0' and nozzle.

I found with 5 squares level/height tests that my nozzle has to be closer than 0.1mm that one would normally set with paper gap.  Otherwise the stripes in the squares are too far apart.

albertoabrao, 2022-11-08:
After much tweaking, so far, the best settings for my S1 with the F4 board are the following:

1) do not use more than 4x4 for mesh points.
2) on Control > Motion > Speed, use the following values (taken from stock firmware):
- Max X Speed: 300
- Max Y Speed: 300
- Max Z Speed: 5
- Max E Speed: 25
- Save settings, turn printer off, then on again, just for good measure.
3) To measure Z offset, I usually do the following.
- First, remove the rubber nozzle protector.
- Now, warm up the hot end - and only that - to 240-ish. Unload filament (if needed), and use a brass brush to clean the nozzle really, really well.
- Let it cool to room temperature. Grab a snack or something.
- **The rest should be run with the bed and nozzle COLD**.
- Clean the bed really, really well. If using the spring steel bed that came with the S1, spray some isopropyl alcohol (70% or 99%, if you can find the latter) on it, rub with a clean toothbrush, then wipe with microfiber towel.
- Now run the tramming wizard until it is level. You don't need to be OCD about it, as long as it passes, it's OK. Ask me how I know... :)
- Get a piece of "Wax Paper", the one you line your trays with when baking. It's just the right thickness, and does not absorb moisture, thus giving a very consistent "drag" every single time.
- Now, Move Z to Home. It will probe the middle of the bed - twice. 
- *Just before it goes down after the second probe*, slide the wax paper between nozzle and bed. Then adjust. You are aiming for a *confident* bite of the nozzle on the paper... you're able to slide it back and forth, yes, but you can feel it's there. The feeling is slightly different for glass and the spring steel flexible bed that came with the printer. Aim for a "confident" bite and you'll get the gist of it after a few tries.
- Now, before you call it quits, remove the paper, hit Move Z to Home again, and make sure it's somewhat the same "feel" for the drag.
- If you do that twice - or thrice - you'll find that either a specific tiny range - say, 3.54 to 3.56 - or even a consistent number - 3.55? - is where you get satisfying results.

Go ahead and test! For PLA, that's all you need to do. For PETG, I add 0.02 extra Z offset. That's it.

If you're lazy like I am, you can install two Cura plugins that will allow you to not have to add the extra Z offset between PETG and PLA:
- Z Offset Plugin
- Material Setting Plugin

After you install these from the Marketplace, just go to Settings -> Materials, then select the pane "Print Settings". On the bottom left corner, there's now a button called "Select settings". Select all the ones you'd like to customize for each material - e.g. brand of filament etc. One of the options I use is Z Offset, then I just add the 0.02 extra that I use for PETG filaments. No need to change every time - or should I say, forget every time? :)

Let me know how it works!

albertoabrao, 2022-11-08:
Oh, one more thing.

**AFTER** you do all of the above, before each print:

Heat bed to temperature, then run tramming wizard. 

If the material requires a different bed temperature (e.g. PLA vs PETG), heat it to the temperature you'll use, then run Tramming Wizard.

That is especially true for the spring steel bed. Glass tends to be fairly consistent, you can usually get away with levelling it cold and it won't change much. 

Spring Steel, on the other hand, varies wildly: level it cold, and when you have it at 75C it'll be completely out of whack...

...at least that's the case with my printer. :)

mhaas, 2022-11-10:
I did some experiments this morning and I thought I would share the raw results. I haven't had time yet to analyze.

I tried all combinations of the following:

* Z Offset 0, Z Offset -3.55
* HS mode on / off
* MULTIPLE_PROBING on/off

Results:

#### Professional Firmware

##### z_offset: 0.0, highspeed_mode: 0

Regular:

```
-0.09	-0.09	-0.07	0.02	0.05
0.10	0.05	0.00	0.04	0.04
0.04	0.03	-0.03	0.04	0.06
-0.02	-0.06	-0.11	-0.04	-0.02
-0.02	-0.09	-0.18	-0.13	-0.14
```

Normalized:

```
-0.07	-0.06	-0.05	0.04	0.08
0.13	0.08	0.02	0.07	0.06
0.07	0.05	-0.00	0.06	0.09
0.00	-0.03	-0.08	-0.01	0.00
0.01	-0.07	-0.16	-0.11	-0.11
```

##### z_offset: -3.55, highspeed_mode: 0

```
-0.11	-0.10	-0.07	0.01	0.05
0.11	0.05	-0.00	0.04	0.02
0.03	0.02	-0.03	0.03	0.06
-0.03	-0.04	-0.10	-0.03	-0.01
-0.03	-0.11	-0.19	-0.15	-0.15
```

Normalized:

```
-0.08	-0.07	-0.04	0.04	0.08
0.13	0.08	0.03	0.06	0.05
0.06	0.05	-0.00	0.06	0.09
0.00	-0.01	-0.07	-0.00	0.02
-0.00	-0.08	-0.16	-0.12	-0.13
```

##### z_offset: 0.0, highspeed_mode: 1

```
-0.11	-0.10	-0.07	0.00	0.06
0.09	0.04	-0.01	0.02	0.03
0.02	0.01	-0.04	0.02	0.06
-0.04	-0.05	-0.10	-0.03	0.00
-0.04	-0.12	-0.20	-0.16	-0.16
```

Normalized:

```
-0.08	-0.07	-0.04	0.04	0.09
0.12	0.07	0.03	0.06	0.06
0.05	0.05	0.00	0.06	0.09
-0.00	-0.01	-0.07	0.00	0.04
-0.00	-0.09	-0.17	-0.12	-0.12
```

##### z_offset: -3.55, highspeed_mode: 1

```
 -0.12	-0.11	-0.08	-0.00	0.06
0.07	0.02	-0.02	0.01	0.02
0.01	-0.00	-0.04	0.01	0.04
-0.05	-0.06	-0.10	-0.03	0.01
-0.05	-0.13	-0.21	-0.17	-0.16
```

Normalized:

```
-0.08	-0.07	-0.03	0.04	0.10
0.12	0.07	0.03	0.06	0.06
0.05	0.04	0.00	0.05	0.09
-0.01	-0.01	-0.06	0.01	0.05
-0.01	-0.09	-0.17	-0.13	-0.12
```

#### Professional Firmware, no MULTIPLE_PROBING

##### z_offset: 0.0, highspeed_mode: 0

```
-0.12	-0.11	-0.09	-0.01	0.04
0.04	0.02	-0.02	0.02	0.02
0.00	-0.00	-0.05	0.01	0.04
-0.08	-0.07	-0.11	-0.02	0.01
-0.07	-0.13	-0.21	-0.16	-0.16
```

Normalized:

```
-0.07	-0.06	-0.04	0.04	0.09
0.09	0.07	0.03	0.07	0.07
0.05	0.05	-0.00	0.06	0.09
-0.03	-0.02	-0.06	0.02	0.06
-0.02	-0.08	-0.16	-0.12	-0.11
```

##### z_offset: -3.55, highspeed_mode: 0

```
-0.12	-0.09	-0.07	0.00	0.06
0.03	0.02	-0.02	0.02	0.04
-0.01	-0.01	-0.05	0.01	0.06
-0.07	-0.06	-0.10	-0.03	0.01
-0.08	-0.13	-0.21	-0.17	-0.16
```

Normalized:

```
-0.07	-0.05	-0.03	0.05	0.10
0.07	0.06	0.03	0.07	0.09
0.03	0.04	-0.00	0.06	0.10
-0.02	-0.01	-0.06	0.02	0.06
-0.03	-0.09	-0.17	-0.13	-0.11
```

##### z_offset: 0.0, highspeed_mode: 1

```
-0.12	-0.09	-0.06	0.00	0.05
0.02	0.01	-0.01	0.02	0.04
-0.01	-0.00	-0.04	0.00	0.06
-0.07	-0.06	-0.10	-0.04	-0.00
-0.09	-0.13	-0.20	-0.16	-0.14
````

Normalized:

```
-0.07	-0.04	-0.02	0.05	0.10
0.07	0.06	0.03	0.07	0.09
0.03	0.04	0.00	0.05	0.10
-0.02	-0.01	-0.06	0.00	0.04
-0.04	-0.09	-0.15	-0.12	-0.09
````

##### z_offset: -3.55, highspeed_mode: 1

````
-0.12	-0.09	-0.06	-0.01	0.05
0.02	0.01	-0.02	0.00	0.04
-0.02	-0.01	-0.05	-0.01	0.06
-0.06	-0.06	-0.11	-0.08	-0.03
-0.09	-0.13	-0.18	-0.15	-0.11
````

Normalized:

````
-0.07	-0.04	-0.01	0.04	0.10
0.07	0.06	0.03	0.05	0.08
0.03	0.04	0.00	0.04	0.11
-0.01	-0.01	-0.06	-0.03	0.02
-0.04	-0.08	-0.13	-0.10	-0.06
````

gcode: https://gist.github.com/mhaas/a1e9866803bc87815e1bee93b12346c2

mhaas, 2022-11-11:
I compared the meshs I generated above to understand which setting has the biggest influence. I compared both the average difference between the meshes as well as the biggest absolute (single-cell) difference. Here's my results: https://github.com/mhaas/s1-ubl-debug/blob/main/mesh.ipynb - scroll down to the diagrams.

Overall, it seems that indeed the MULTIPLE_PROBING has the biggest influence by far.

On the weekend, I will manually generate a mesh as a ground truth and see which of the different settings comes closest.

Additional things one might investigate based on the data:

- Influence of z-offset?
- comparing normalized meshs instead of absolute meshs
- comparison to ground truth

mhaas, 2022-11-23:
I spent a bit more time looking into this. On my printer, even at Z offset 0, there is a marked difference between Multiple Probing enabled/disabled.

[image]

I am collecting all information in a wiki: https://github.com/mhaas/ender-3-s1/wiki/Automatic-Bed-Leveling-Issues

You are cordially invited to contribute, edit, update etc.

My current hypothesis is that the Z probe feedrate has an impact on the probe reliability: https://github.com/mhaas/ender-3-s1/wiki/Automatic-Bed-Leveling-Issues#the-z_probe_feedrate-hypothesis

I don't have access to my printer right now. I think the next step will be to get the devug leveling output (see second link). If you can help out here that would be great!

mriscoc, 2022-11-23:
Be aware that BLTouch and printer mechanical repeatability is no better of +-0.05 mm, so the difference falls within the error margin.
M48 standard deviation is not the real accuracy more information about repeatability gives the **range** value.

mhaas, 2022-11-23:
That's a good point. It is even possible that my finding that multiple probing has a big impact on the mesh is exactly because MP gives more accurate results. Still, the patterns that I see are between MP on and off do not seem to be enrirely random.

Going forward, I will include a test print in my experiments to understand if a given configuration has a positive or negative impact. Of course that is harder to automate.

By the way, I earlier found an issue in the Klipper repo indicating that BLtouch results depend on Z speed. Of course, that is a completely different firmware and architecture.

grrmisfit, 2022-11-24:
something is off, doesnt matter what settings the mesh is wrong. I even swapped probes to verify it wasnt a probe issue. corners are usually off on the left side. i can clearly make a mesh and watch it be off when printing a test print. i manually level with paper and wizard tells me im off but when i adjust and go back to manual the height is clearly way off. short of going to klipper i dont know what to do.

hayden-t, 2022-11-24:
@grrmisfit please see my above post https://github.com/mriscoc/Ender3V2S1/issues/195#issuecomment-1221262298

grrmisfit, 2022-11-24:
> @grrmisfit please see my above post [#195 (comment)](https://github.com/mriscoc/Ender3V2S1/issues/195#issuecomment-1221262298)

Read it, tried it, still major variance on 2 left corners. I even disabled rods and made sure x gantry was level on both sides.  Test print has left side grinding bed and right side sticking to barely sticking. I'm going to do a test print without using mesh and see the results. I've taken this printer apart multiple times and center prints are hit and miss with outside being non existent without wheel adjustment

hayden-t, 2022-11-24:
so you made a mesh with the probe at 0 offset ? and then saved the mesh and the lines under your z are blue when printing ?

grrmisfit, 2022-11-24:
> so you made a mesh with the probe at 0 offset ? and then saved the mesh and the lines under your z are blue when printing ?

Yes, and if you go manual tramming vs wizard the wizard is always wrong. If I level the corners via wizard then test using manual it'd to tight on one side and to high on others

hayden-t, 2022-11-24:
so you are doing manual or wizard tramming with the probe mounted at 0 with my mount and have the offset in settings to 0 too ?

KBalage, 2022-11-25:
I have an Ender3 S1 Plus with F4. I'm creating a mesh, tested it with HS on/off, MP to 2, then 1, doesn't matter:
I can see the mesh in the View Mesh menu, it's maybe valid, maybe not (I think it's correct).

But my feeling is that it's simply doesn't move the steppers to adapt to the stored mesh during printing of the 1st layer.

I have watched very closely my stepper rods during a bigger print and while I had the Z blinking (with blue lines) and the Z height was changing on the display - yet I haven't seen the Z rods moving during the whole 1st layer, only for Z hops during retraction.

mhaas, 2022-11-26:
There is a new hypothesis: Electromagnetic Interference. Details here: https://github.com/mhaas/ender-3-s1/wiki/Automatic-Bed-Leveling-Issues#the-emi-hypothesis

[image]

A possible worksround is to build with PROBING_ESTEPPERS_OFF.

Would be great if someone can try this!

KBalage, 2022-11-26:
I will test the PROBING_ESTEPPERS_OFF. And I have ordered an SKRME3V3 too...

KBalage, 2022-11-27:
So I've tested the PROBING_ESTEPPERS_OFF and not convinced... I still can see the same mesh data regardless of using HT or multiple probing or any of these combinations. I had a weird issue after recompiling the firmware - the printer started with High temp - printer killed (after cold start), but then recompiling and updating it again it was ok. 

I've probably have a bed with a little defect, the front and the back part of the bed are much higher than the middle for some reason.
[image]

I'm using M420 in my starting code, Z is blinking, the mesh view shows almost the same values, yet the first layer is not perfect, too low on same areas, too high on others, like the mesh simply doesn't taken into consideration or I have a completely different bed in reality than it was recognized during the levelling procedure.

Waiting for the SKRME3V3 ... thinking on Bed Distance Sensor...

mhaas, 2022-11-27:
I also did some tests today with  `PROBING_ESTEPPERS_OFF` and/or `PROBING_STEPPERS_OFF`

Results here: https://github.com/mhaas/ender-3-s1/wiki/Automatic-Bed-Leveling-Issues#testing

I don't think this is the big improvement I have been looking for :)

Snolte1001, 2022-11-29:
Hello, from me is the post on Facebook ! 
During my tests, I had the cable of the CR/BL touch completely laid separately, it was so that the X stepper also caused interference, so the deactivation of the extruder stepper can not necessarily lead to hoped for success.  To deactivate the X stepper during the measurement is rather problematic ;-)

KBalage, 2022-12-02:
I had a gantry leveling problem - the right part of the X axis was ~0.4 mm lower than the left part. 
I can see this happening even on my picture above in this thread btw, what I don't understand is why the first and last Y line doesn't affected, and why I have the 1st and last line much higher than the rest of the bed. Nevertheless I've fixed the X axis, because that was surely wrong. I've also downloaded and using the UBL version for F4 and it's working a bit better - I'll test printing on different areas of the bed, but at least I can see now that the Z rods are doing some job during a bigger print.

MarkforU, 2022-12-02:
I put timing belts on my z-rods, which I run 2 machines. You won't have the problem with height control, and yes UBL does help, but getting your x-gantry straight if running duel z-rods you'll fight that forever on the right side being higher or lower until you put a timing belt on it. I've been going 2 weeks Non stop on 2 printers knocking out Chinese Flexi Dragons that sit up and have 67 parts that finish as one. Fidget spinners and stars for 18 grandchildren. For the dragon I do use the 9x9 mesh even though they say 5x5 is enough. Keep printing 24/7 off original mesh using UBL...
[image]

grrmisfit, 2022-12-14:
To add to this, I created a 4x4 mesh, then went back to edit mesh and points are way off. Take point 2-0, my manual edit points value at .07 while probe says -.7 should be my value. I get to the back and I can't even go down enough.  So it's something and the probe is basically useless at this point. Might as well do manual mesh.

[24 more replies]

closed 2023-03-13 by mriscoc