mriscoc/Ender3V2S1 #196: BUG on Start New Print.

pacocatech, 2022-06-11:
When I click print, the firmware update message appears.

See the video, I'm using the version with BLTOUCH.

https://drive.google.com/file/d/1hpIaxUF9s5XLgrc3daU9g7BYnBe1wQWA/view?usp=sharing

My board is 4.2.2

mriscoc, 2022-06-12:
The video doesn't show your bug.

pacocatech, 2022-06-12:
Sorry, i posted wrong video, this is correct: https://drive.google.com/file/d/18cDZCn82P1qszG-3-w0zkJFHTdvf-tk4/view

locksen-nl, 2022-06-13:
I have the same problem also 4.2.2 with a bltouch

mriscoc, 2022-06-13:
Seems to be a problem with some kind of memory cards, try with other SD card, format to FAT32, put no more of 20 files on it.

pacocatech, 2022-06-14:
I even thought it could be the card, I got a new class 10 card, formatted it as FAT32 because NTFS doesn't work and just added the firmware and then formatted it again and left a maximum of 4 files and the problem still persists.
In the old firmware version this does not happen.

mriscoc, 2022-06-16:
Please, can you specify what precompiled version you are using?

pacocatech, 2022-06-20:
this version: [Ender3V2-422-BLT-20220527.bin](https://github.com/mriscoc/Ender3V2S1/releases/download/20220527/Ender3V2-422-BLT-20220527.bin)

esrpradeep, 2022-06-24:
I have the same problem after upgrading to Ender3V2-427-BLT-20220527.bin. Remove the 'SDCard' as a workaround and press 'Print' - ignore the 'No Media' message. While still in the 'Print' menu screen, insert the 'SDcard' to select and print the desired file.

faucherc, 2022-06-27:
I have the same problem after upgrading to [Ender3V2-422-BLT-20220527.bin](https://github.com/mriscoc/Ender3V2S1/releases/download/20220527/Ender3V2-422-BLT-20220527.bin).  I have a 4.3.1 mainboard and a DACAI display.

I have tried with 2 different SD Card and I have seen the problem with both of them.  I never encontered this kind of problem with these 2 cards with the Creality stock firmware.

When the problem occurs, generally, it occurs only one time instead of many time like in pacocatech video's.

mriscoc, 2022-07-17:
Please verify with latest version.

Moobien, 2022-07-17:
This was the same Issue I reported a few weeks back.  Card was completely fine.

esrpradeep, 2022-07-17:
> I have the same problem after upgrading to Ender3V2-427-BLT-20220527.bin. Remove the 'SDCard' as a workaround and press 'Print' - ignore the 'No Media' message. While still in the 'Print' menu screen, insert the 'SDcard' to select and print the desired file.

Update: Upgraded to Ender3V2-427-BLT-20220716.bin, and I haven't observed the 'SDCard' issue so far.

Observation from my end: 

JULY2022 release has the SDSORT_LIMIT as '40' in the file 'Configuration_adv.h' compared to May2022 release that had '60' for Ender3V2-427-BLT under the link 'Special_Configurations'.

Latest 'Configuration_adv.h' from Special configurations
#define SDSORT_LIMIT       40     // Maximum number of sorted items (10-256). Costs 27 bytes each.  // MRiscoC Increase number of sorted items

I'm not sure if this was the root of the problem or could have been resolved due to the latest Marlin bugfix branch.

mriscoc, 2022-07-17:
SDSORT_LIMIT is only a limit value, I reduced it for cases where the SD card has a large amount of files and the memory limit could be exceded.
I think that the error could comming from the Marlin core.

mriscoc, 2022-08-17:
Print File menu list has been improved in the last version, please test.

mriscoc, 2022-08-22:
No more user feedback.

closed 2022-08-22 by mriscoc