mriscoc/Ender3V2S1 #198: mesh not being applied ender  3 s1 with f4 chip

dasdv124, 2022-06-15:
no idea what i am doing wrong, but it seems like the created mesh is not being used. no matter what i do the middle is always too high and the the corners too low.  this is my g-code 

G28 ;Home
M420 S1 Z2 ;Use mesh level upto 2 mm
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up

the steps i do : 
set z offset with paper, auto home then manual bed trimming,mesh create

[image]

webmogul1, 2022-06-15:
Do the 2 bars below Z change color when you are printing?
[image]

dasdv124, 2022-06-15:
yes it does change color to blue when printing

Victordiazbis, 2022-06-17:
Have you edit manually the mesh to check if it makes any difference?

dasdv124, 2022-06-17:
> Have you edit manually the mesh to check if it makes any difference?

I have not, what's a setting I should change to test it ? Thanks

neumo, 2022-06-18:
> yes it does change color to blue when printing

my S1 changes the color to red!? What does the color red or blue mean?

mriscoc, 2022-06-18:
> my S1 changes the color to red!? What does the color red or blue mean?

https://github.com/mriscoc/Ender3V2S1/wiki/3D-BLTouch#enable-mesh-level-compensation

cybersatanizer, 2022-07-09:
Hi, i also have problem with mesh leveling. i did:
-bed tramming
-z offset
- mesh leveling

then use G29 at start script, and trying to make first layer calibration from superslicer and lines under Z is going from white to blue but center is perfect, back left too high, front right too close, like printer did nothing about g29?

mriscoc, 2022-07-17:
Currently some S1 (F1 and F4) have issues with the CRTouch, sadly it seems to be a hardware problem. I suggest you use the automatic leveling as reference and then do a mesh editing and save to two slots (one for backup). That will give you the best results.

closed 2022-07-17 by mriscoc