mriscoc/Ender3V2S1 #199: Z Axis E-Steps

I-wish-2, 2022-06-18:
**Is your feature request related to a problem? Please describe.**
This is not related to any problem, it is just an enhancement to the Z axis e-steps when up grading the leadscrew to a finer 2mm pitch.

**Describe the solution you'd like**
I have a strong desire to change my stock Z axis lead screw for one that is a 2mm Pitch, 1 Start & 2mm Lead in addition to adding a Anti Backlash Spring Loaded Nut to the lead screw. This is in an effort to stop the Z axis drop and allow for a finer print quality.  This lead screw will require the Z axis to have an E-step value of about 1600.  Currently the max E-step value allowed is 999.9.  I'm not sure if the decimal place will be required for this new lead screw.
Please note that your firmware is loaded on to my Ender 5 Pro for for the 4.2.2 motherboard and display.  The original display was replaced with a E3 V2 display v 1.4 and your private firmware.  This E5 Pro is fully functional in all aspects after quite a bit of elbow grease was applied.
Your firmware is loaded on my E3 V2 as well which will also have its lead screw upgraded.

**Describe alternatives you've considered**
I have not looked into any other options regarding the Leadscrew upgrade other than to keep using the original stock one.  Which is an 8 pitch or 8mm of movement for 360 degrees of rotation.  The other is 2mm of movement for 360 degrees of rotation.

**Additional context**
Here is a youtube video on the lead screw up grade if you need it.  https://www.youtube.com/watch?v=lBpXHWJG7aM
Thank you Sir for a fine effort with your firmware and share.  I came here through 3DHub.   Sorry for being wordy.
Doug

I-wish-2, 2022-07-07:
Greetings Miguel,
Thank you very much for your reply and great news.  This will be quite an improvement for the step resolution of the Z axis on Ender 3 V2 and Ender 5 Pro that I have running your firmware.
Do you feel that it will require a motherboard and display firmware update?  Thank you once again.
My Best Regards
Doug

mriscoc, 2022-07-08:
You will need to update only the main firmware.

I-wish-2, 2022-07-08:
I hope the testing goes well.  Once again thank you sir.
Best Regards
Doug

I-wish-2, 2022-07-10:
Greetings,  I wanted to add that if you would like a test site using an Ender 3 V2 and a Ender 5 Pro my machines are available.  Both of these printers are running your current firmware and are working well.  If you have an interest please let me know.  I'm currently retired and have the time to devote to such a project.
Best Regards
Doug

mriscoc, 2022-07-10:
> Greetings, I wanted to add that if you would like a test site using an Ender 3 V2 and a Ender 5 Pro my machines are available. Both of these printers are running your current firmware and are [...]

Thank you.

mriscoc, 2022-07-16:
Maximum limit of E-steps have been increased in the latest release.

closed 2022-07-16 by mriscoc

I-wish-2, 2022-07-18:
Greetings Miguel,  Thank you for the E-Steps changes.  I will start the lead screw upgrade on the Ender 5 Pro today and can post the results here in this comment area.  Again thank you for your efforts.
Best Regards
Doc'