mriscoc/Ender3V2S1 #200: Add a longer time before cooling down during M600 filament change

ecogice, 2022-06-18:
**Is your feature request related to a problem? Please describe.**
I set up M600 filament changes in my gcode often and the screen says "Click to heat" and starts cooling down within about a minute after the command

**Describe the solution you'd like**
I'd like the time before cooling down to be changed to 2-3 minutes

mriscoc, 2022-06-19:
I don't remember the current timeout but it is more than 1 minute, remember that more time is more risk to have crystallization of the residues inside of the nozzle and that could produce clogs.

ecogice, 2022-06-20:
I used a stopwatch and from the time that it raises off the build plate to the time that it says "Click to heat," it was about a minute and 5 seconds. I do understand that leaving it heated without printing can cause clogs, but adding 60 seconds more to the waiting time would make changing filament faster

mriscoc, 2022-06-20:
OK, I will check, you could help me by finding the relevant parameter in the configuration files.

mriscoc, 2022-06-20:
Excellent, I will increase PAUSE_PARK_NOZZLE_TIMEOUT to 90 in the next version. Thanks.

cristianmur, 2022-06-22:
una pregunta de novato, no se podria hacer una retraccion mas larga en el cambio de color, para no tener que estar tan pendiente de la impresora?

yo lo digo por que ya me ha pasado varias veces que por algunproblema como filamento roto, se me a quedado parada la impresion toda la noche y luego sigue genial.

La verdad que estoy enamorado de tu firmware es increible, al principio lo instale pero al tener el chip F4 que era nuevo no guardaba en la eprom, pero ya lo solucionaste. No me acordaba y estaba trabajando con un marlin externo hasta que ayer me dijeron que ya lo habias solucionado lo de la placa nueva.
Muchas gracias , eres un genio pedazo de compilacion.

mriscoc, 2022-06-23:
> una pregunta de novato, no se podria hacer una retraccion mas larga en el cambio de color, para no tener que estar tan pendiente de la impresora?
 
La longitud de carga y descarga puedes manejarla desde el menu `Advanced / Filament Settings`

ZoeESummers, 2022-07-02:
> Excellent, I will increase PAUSE_PARK_NOZZLE_TIMEOUT to 90 in the next version. Thanks.

Can it be changed to resume on touching a button on the screen, like you do on the Prusa printers? So no fixed timeout is needed and the screen will show the temps and will resume when back up to temp and continue has been touched.

mriscoc, 2022-07-05:
> Can it be changed to resume on touching a button on the screen, like you do on the Prusa printers? So no fixed timeout is needed and the screen will show the temps and will resume when back up to [...]

It could be dangerous not to implement an automatic timeout.

mriscoc, 2022-07-16:
PAUSE_PARK_NOZZLE_TIMEOUT  increased to 90s in the latest version

closed 2022-07-16 by mriscoc