mriscoc/Ender3V2S1 #202: Eextrusion temperature

fabiomaticus, 2022-06-22:
**Describe the bug**
In the advanced menu, filament setting, I set the extrusion temperature to 205 degrees. After each print, the value returns to 180.

**To Reproduce**
1. Go to Advanced menu
2. Click on Filament setting'
3. Scroll down to Extruder min. temp'
4. value is 180; set to 205 anda save
5. after print, the valure return to 180

**Version (please complete the following information):**
Firmware version and release date
20220527 version

mriscoc, 2022-07-06:
No reproducible. Check if your G-code file has a M502 or M302, both G-codes resets the value that you configure.

closed 2022-07-06 by mriscoc

fabiomaticus, 2022-07-06:
This is my g-code:
M501; Read values from EEPROM
M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z20.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 T0 S93.00

G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S150 ; set temporary nozzle temp to prevent oozing during homing and auto bed leveling
M140 S[first_layer_bed_temperature] ; set final bed temp
G4 S10 ; allow partial nozzle warmup
G28 ; home all axis
M420 S1 Z3 ; Use mesh level upto 3 mm
G1 Z50 F240
G1 X2 Y10 F3000
M104 S{first_layer_temperature[0]} ; set final nozzle temp
M190 S[first_layer_bed_temperature] ; wait for bed temp to stabilize
M109 S{first_layer_temperature[0]} ; wait for nozzle temp to stabilize
G1 Z0.28 F240
G92 E0
G1 Y140 E10 F1500 ; prime the nozzle
G1 X2.3 F5000
G92 E0
G1 Y10 E10 F1200 ; prime the nozzle
G92 E0