mriscoc/Ender3V2S1 #207: Thumbnail image script error with Prusa 2.4.2

CJ65CM, 2022-06-26:
Let me start by apologizing if it is improper for me to have started a new issue here for something that has already been sorta covered (pretty new to this and wasn't sure if I should have commented in the past "issue" that has been closed or just start a new one as I am).

I am having the exact same script error issue as seen in this one:
https://github.com/mriscoc/Ender3V2S1/issues/174

This is with all of the latest software as of last night. Since this was an issue that was supposedly resolved with the Python script being updated to support 2.4.2 and kind of think I might be doing something wrong. I'm not the most savvy person when it comes to some of this computer and coding stuff but often can fumble my way to resolving issues that are errors on my part but not this time so far. 

I've tried both the latest version of Python (3.10.5) with no luck and have and have tried as far back as 3.0.1 (with a few others in between) and all give me the same error so I dont thing its that.

I also have tried saving the CreateHeaderAndThumbnail.py file into several different directories including creating a folder called "scripts" in the "PrusaSlicer" like the example here shows mricoc's having:
https://raw.githubusercontent.com/mriscoc/Marlin_Ender3v2/Ender3v2-Released/slicer%20scripts/prusa-superslicer/prusasettings3.png

mine didn't have that "scripts" subfolder so I created it and trying storing the script there too but no luck with that either.

Could there be something else I am missing or is it still an issue that wasn't resolved?

Thank you in advance,
Chris

Double077, 2022-06-29:
mee too same issue exactly!

CJ65CM, 2022-06-30:
I wonder if the download (right click and save link as) is a problem because if I open the python file that is downloaded in Notepad ++, it is a TOTALLY DIFFERENT "code" than what you see if you just left click on the text that you are supposed to "right click and save as" (which takes you the raw code that actually looks like a typical Python to the best of my knowledge).

I tried to copying all the text (code) that is on the page that opens if you just left click the Prusa/Super Slicer hyper link and creating the CreateHeader.....py file and trying that and it gives the same error (if memory serves me correctly). So at this point, I've done about all the "fumbling my way to a resolution" that my limited knowledge and understanding of all this allows with no success.

Hopefully Miguel see (or has seen) this and can resolve it.

mriscoc, 2022-06-30:
Tested and it is working good with this version:
[image]

Open your script file, maybe it is corrupt, open a terminal and confirm that your Python installation is ok:
```
>python -V
Python 3.10.4
```
Check if you have Pillow installed:
```
>pip3 list Pillow
Package Version
------- -------
Pillow  9.1.1
```

CJ65CM, 2022-06-30:
THANK YOU!!!!!! Miguel

The problem was Pillow wasn't installed. Working as intended now.

CJ65CM, 2022-06-30:
Also, I'm not sure if its just my case or what but when I "Right Click and Save Link As" the Prusa/Super Slicer hyperlink as the wiki say to do, the .py file that downloads was not the actual code file (even though it is named as though it is).
What I did was copied the code (same as what you posted and pasted it in a blank Notepad ++ tab and then saved it with the correct name .py

Any idea why that would be like that? Could that have something to do with the fact that I didn't (at the time) even have Python program on the PC or anything to do with that or maybe just a hyperlink problem with Github?

What was shown in the text editor when I opened it was basically just a bunch of HTML code (over 3000 lines of it to be exact).

mriscoc, 2022-06-30:
> Any idea why that would be like that? Could that have something to do with the fact that I didn't (at the time) even have Python program on the PC or anything to do with that or maybe just a hyperlink problem with Github?

Maybe it depends on the browser version and/or operating system. I test with Chrome and Edge on Windows.

CJ65CM, 2022-06-30:
I just tried it again and what it downloads is the right file so I'm guessing it was either just a temporary glitch and I just had bad timing on my part or it had something to do with not having python on my PC at the time of downloading.... who knows...

closed 2022-06-30 by mriscoc