mriscoc/Ender3V2S1 #208: Purple Theme

MrBamse, 2022-06-26:
Made a purple theme file, feel free to use it and add it to the list of example themes. :)

;Purple Theme File
;
C11 E2  R15 G0 B14      ;Screen Background
C11 E3  R30 G0 B30      ;Cursor
C11 E4  R0 G0 B0          ;Title Background
C11 E5  R31 G63 B31    ;Title Text
C11 E6  R31 G63 B31    ;Text
C11 E7  R30 G0 B30      ;Selected
C11 E8  R20 G0 B20      ;Split Line
C11 E9  R30 G0 B30      ;Highlight
C11 E10 R20 G0 B20     ;Status Background
C11 E11 R31 G63 B31   ;Status Text
C11 E12 R6 G15 B8       ;Popup Background
C11 E13 R26 G53 B26   ;Popup Text
C11 E14 R30 G0 B15     ;Alert Background
C11 E15 R31 G56 B15   ;Alert Text
C11 E16 R31 G63 B31   ;Percent Text
C11 E17 R30 G0 B30     ;Bar Fill
C11 E18 R31 G63 B31   ;Indicator value
C11 E19 R31 G63 B31   ;Coordinate value
;
C11 E0  ;Preview changes
;M500 ;Save to EEPROM
;M997 ;Reboot

mriscoc, 2022-06-26:
Great! Thank you.

MrBamse, 2022-06-27:
> Great! Thank you. Posted here: https://github.com/mriscoc/Ender3V2S1/wiki/Color-Themes#mrbamse-purple-theme

Cool, thanks dude!
Btw, you're doing a great job! :)

closed 2022-06-27 by mriscoc