mriscoc/Ender3V2S1 #210: Extreme Over Extrusion

VaioHaZZard, 2022-06-27:
**Describe the bug**
During a print the nozzle would slow almost to a stop (100mm/s down to 20mm/s) and the extruder would continue to extrude at the normal rate causing a massive build up of filament and extreme skipping of e-steps.  On the next layer the nozzle would run into the hard blob of plastic and shift the print causing a failure.

**To Reproduce**
Steps to reproduce the behavior:
This issue occured in random X,Y, and Z locations on 10 prints in a row.  Could not eliminate it by changing settings on machine.  Could not reproduce error in same spot consecutively.  

**Version (please complete the following information):**
Firmware version [Ender3V2-422-BLT-20220527] and [Ender3V2-422-BLTUBL-20220527] 

**Additional context**
Troubleshooting included: switching from Cura 5 back to Cura 4.  Switching to a different formatted memory card.  Changing from the UBLBLT to the BLT version of the 20220527 firmware.  Changed number of mesh points from 6X6 to 4X4.  Reflashed both versions of 20220527 several times.  Ensure power failure was off.   Would also get frequent memory card read errors and machine would reboot frequently when memory card was inserted. 
Reverting back to Ender3V2-422-BLTUBL-20220419 solved the issue.  
I have used every version of your firmware to date and 20220527 was the only one with this issue.  
Thank you for your hard work and I look forward to your next release.

mriscoc, 2022-07-05:
Thank you, seems to be some bug with certain SD memory cards. I don't have that problem with any of my memory cards. Unfortunately it can't be fixable by the UI, it could be a problem with the Marlin core, I hope that next version based in the latest Marlin sources can fix the bug. Try to have no too much files on the memory card and also use cards lower or equal to 32GB.

mriscoc, 2022-07-05:
Could be related to: https://github.com/mriscoc/Ender3V2S1/issues/196

SorenRR, 2022-07-05:
Standard firmware Ender3V2-422-MM-20220527. 
I'm using the card that came with the printer. Netac 8GB MicroSD HC. Two folders with a few test print for 0.4 and 0.6 nozzle.
Inserting the card into a freshly booted printer is not a problem. Removing the card will show "Media Init Fail".
IF Power Outage is selected the extruder will stop after a while but "print" will complete.
Powercycling the printer with card inserted clears the error and print will succeed regardless if Power Outage is selected.

mriscoc, 2022-07-05:
> Standard firmware Ender3V2-422-MM-20220527. I'm using the card that came with the printer. Netac 8GB MicroSD HC. Two folders with a few test print for 0.4 and 0.6 nozzle. Inserting the card into a [...]

The included memory card is not always of the best quality, could be better to get a new SD.

mriscoc, 2022-08-17:
After many test, it was not possible to replicate the issue.

closed 2022-08-17 by mriscoc