mriscoc/Ender3V2S1 #215: Ender 3 S1 Expansion Interface Software Control

jaredbiehler, 2022-07-05:
This is a shot in the dark (ha) - I'm wondering if the controller has any ability to interface with or control the Expansion Interface that's included on the Ender 3 S1 models? The most popular accessory I've seen this used for is a light bar and it'd be useful to allow this light be controlled (especially remotely). I thought I'd ask here before setting up a relay with OctoPrint. Thank you for your time and all the wonderful features in this firmware!
[image]

mriscoc, 2022-07-16:
The port can be controlled but currently there is not a device that use it.

jaredbiehler, 2022-07-17:
https://www.creality3dofficial.com/products/ender-3s1-led-kit does. I was hoping to figure out a way to turn this light on / off via this port remotely via OctoPrint through the controller, if possible.

mriscoc, 2022-07-17:
> https://www.creality3dofficial.com/products/ender-3s1-led-kit does. I was hoping to figure out a way to turn this light on / off via this port remotely via OctoPrint through the controller, if possible.

I think that the lamp is not using the expansion port (PWM) but only the 24V power supply.

Lungenpest, 2022-07-22:
> > https://www.creality3dofficial.com/products/ender-3s1-led-kit does. I was hoping to figure out a way to turn this light on / off via this port remotely via OctoPrint through the controller, if possible.
> 
> I think that the lamp is not using the expansion port (PWM) but only the 24V power supply.

It is connected to the expansion port.

https://www.youtube.com/watch?v=LifRM4gGVi0

[image]

mriscoc, 2022-07-22:
From your video, only power is used.:
[image]

mriscoc, 2022-07-22:
From your video, only power is used:
[image]

jaredbiehler, 2022-07-22:
Ah I was wondering if that's what you were referencing or if you had meant that you believed that light was connected directly to the internal rails of the 24v power supply. So there's only the ability to control the PWM settings, but nothing that can turn the 24v on / off to this expansion interface via software then?

mriscoc, 2022-07-23:
Power supply is not under management of the control board.

closed 2022-07-29 by mriscoc