mriscoc/Ender3V2S1 #218: Custom higher hotend temp(>270°C) triggers thermal runaway warning

BogusF, 2022-07-09:
I'm on a e2s1 pro with a e3v2 knob-display running your firmware and I used the custom Gcode C104 U1 T315 to unlock my full temp range. I couldn't enter the gcode into pronterface, it claimed it didn't recognise the syntax. So I created a gcode file containing:

---
;set new Temp
C104 U1 T315

;save settings
M500

; display for confirmation
M117 Temp setting done!
---

Which initially worked, i confirmed manually in the set temps menu and I could select temps up to the entered range. However, whenever the Hotend actually reaches 270°C or above the printer enters panic mode and wants to be shut down immediately. Apparently, there's a different monitoring routine somewhere, that needs to be updated with the new maximum hotend temp so the safety protocol isn't falsely triggered.

Anyways thanks a LOT for your work on tis firmware!

mriscoc, 2022-07-09:
The absolute maximum is defined by the highest value in the calibration table for a given thermistor, the save maximum is the previous value minus 15 °C of overshoot.

BogusF, 2022-07-10:
Sorry I'm not sure I understand your answer.
So is there a way for me to change values in the calibration table without recompiling? 
With "_previous value_" you mean the highest value in the calibration table? 
That 15K overshoot is why i entered 315°C as the upper limit. Didn't help at all of course.

I thought, the custom gcode is there for users who need higher temps and don't want to/can't recompile. Right now it doesn't serve that purpose since there's another limit at some other place (calibration table, apparently) that stops the show early.

mriscoc, 2022-07-10:
The custom G-code is currently an exceptional way to increase the maximum editable temperature, because it is dangerous I don't create a item in the menu for enable it. But there is another hard limit, the maximum value in the thermistor calibration table. The stock calibration table for the stock thermistor has as maximum 300°C, some other tables allows higher temperatures as the thermistor table for Volcano hotends.

BogusF, 2022-07-11:
Yeah I'm fine with no menu, your safety concerns are valid, I just wanted to mention that pronterface had trouble accepting the custom gcode. (might have been my fault, I didn't try very hard, creating a gcode file is easy enough)

As for the _raise hotend temperature with custom gcode_, maybe specify a range so people don't blindly run into this issue as [they do right now](https://www.reddit.com/r/Ender3S1/comments/vtyg4d/how_to_raise_hotend_temp_limit_in_mriscoc_pro/). Thanks for hinting at the volcano hotends, so that's the way I'm gonna get my high temps back.

BogusF, 2022-07-16:
Thanks for clearing the pronterface issue up!

So if i wanted to have 300°C Hotend temperature, it would be possible to take the calibration table (#13 in this case) and just extend the upper end till 320°C?
I mean value progression in that range is fairly linear at about 86% every 10K, so those two values aren't very hard to come up with and I can test the resulting temperatures with a type-k thermometer afterwards and add corrections if needed. Also, since the "sprite pro" hotend reaches 300°C on stock ROM, it should be safe to run it at those temps. (I'm fine with ruining my hotend in the sake of science btw ;) )
Something like this:

```
[...]constexpr temp_entry_t temptable_14[] PROGMEM = {
  
  { OV( 14,96), 320 },
  { OV( 17,31), 210 },
  { OV( 20.04), 300 },
  { OV( 23.19), 290 },
  { OV( 26.71), 280 },
  { OV( 31.23), 270 },
  { OV( 36.52), 260 },
  { OV( 42.75), 250 },
  { OV( 50.68), 240 },
  { OV( 60.22), 230 },
  { OV( 72.03), 220 },
  { OV( 86.84), 210 },
  { OV(102.79), 200 },
  { OV(124.46), 190 },
  { OV(151.02), 180 },
  { OV(182.86), 170 },
  { OV(220.72), 160 },
  { OV(316.96), 140 },
  { OV(447.17), 120 },
  { OV(590.61), 100 },
  { OV(737.31),  80 },
  { OV(857.77),  60 },
  { OV(939.52),  40 },
  { OV(986.03),  20 },
  { OV(1008.7),   0 },
```

If I looked it up correctly, calibration table assignment for nozzle temperature is done in Configuration.h by "#define TEMP_SENSOR_0 14", would that work if I created the updated table#14? 
I'm using VSCode with PlatformIO and Auto Build Marlin, if I hit build, it offers me two options: 
STM32G0B1RE_btt and STM32G0B1RE_btt_xfer. Is there a meaningful difference? Google didn't help.
I couldn't find flags to choos for F1 or F4 board, but it probably defaults to F1 and that's what I need.
And sorry if I'm asking dumb questions, if those are answered somewhere already, just point me in that direction.

mriscoc, 2022-07-16:
For know if it is important the selected type of thermistor is good to analyze these plots:
[image]
From -15 to 350 °C

[image]
From 150 to 300 °C

The latest pre-compiled firmwares are loaded with a extended Steinhart- Hart regresion table, that allows me to extend the maximum range of operation with better accuracy

Some tables at Marlin are not complete for high temperatures, for example for the thermistor type #5 of the Volcano hotends:

[image]
In grey SteinhartHart regresion, in blue, incomplete stock table

Stock table has a big gap between 300 and 713°C
```cpp
// R25 = 100 kOhm, beta25 = 4267 K, 4.7 kOhm pull-up
// 100k ParCan thermistor (104GT-2)
// ATC Semitec 104GT-2/104NT-4-R025H42G (Used in ParCan)
// Verified by linagee. Source: https://www.mouser.com/datasheet/2/362/semitec%20usa%20corporation_gtthermistor-1202937.pdf
// Calculated using 4.7kohm pullup, voltage divider math, and manufacturer provided temp/resistance
constexpr temp_entry_t temptable_5[] PROGMEM = {
  { OV(   1), 713 },
  { OV(  17), 300 }, // top rating 300C
  { OV(  20), 290 },
...
```
Stock tables are perfectly valid if you print under 285 °C

How to understand this easily? Lets imagine that you are heating your nozzle to  230°C and have a T13 thermistor and your firmware was compiled with a T13 table. So you will read correctly 230°C at screen. But if your firmware was compiled with a T1 table, the real temperature will be 237°C and in the case of T5, the real temperature will be 220 °C. In all cases you will see 230°C in the screen.

mriscoc, 2022-07-16:
Creality gave us mixed answer about of the thermistor type of the extruder pro, the last comment about that is the S1 Pro is using the same thermistor that S1, that is, type 1 (from the official S1 repository)

I extended the table for the S1 version and added the posibitily of increase the temperature range using the custom G-code C104. But only with a calibrated temperature meter or a radiometer we can verify if efectively we can use type 1 with the Sprite pro extruder.

BogusF, 2022-07-17:
I can't say my thermometer is calibrated, but I did a quick sanity check with icewater/boiling water and got 0°C and 100°C as results.  I'd think that's good enough. It has a resolution of 1°C, could be better, but it's ok I guess. 

I tested the temps from 150-300 in steps of 5. I put some thermal grease under the silicone sock and wedged the thermocuple in there, I insulated the lower hotend with rockwool and tinfoil to keep cooling effects at higher temps minimal. Here are the results:

```
set T        measured T         %deviation
300		289		3,7
295		283		4,1
290		280		3,4
285		275		3,5
280		268		4,3
275		264		4
270		260		3,7
265		255		3,8
260		250		3,8
255		245		3,9
250		240		4
245		235		4,1
240		231		3,8
235		226		3,8
230		221		3,9
225		217		3,6
220		212		3,6
215		208		3,3
210		203		3,3
205		198		3,4
200		194		3
195		189		3,1
190		184		3,2
185		179		3,2
180		174		3,3
175		170		2,9
170		165		2,9
165		160		3
160		156		2,5
155		150		3,2
150		145		3,3

```
During the measurements, towards the higher temps, I realized that the PID-algorithm had increasing trouble keeping the temperature on setpoint and started to undulate. I had to anticipate the swinging to read a value at the right time. It got particularly bad at 290°C, then 295° was perfectly stable and 300° it was undulating again. No Idea what that's about.

The fact that resulting temps are lower than what they should be confirms my suspicions, that temps are actually a bit low. I'm usually printing PLA at about 215°C, which would be too high if temps were correct. Of course i can't say how good the thermal coupling between my probe and the hotend actually was, but the grease+insulation helped quite a lot, it got the temp about 3-4K closer.

mriscoc, 2022-07-17:
Could you do the same test but with the precompiled version? for extend the temperature range use C104 U1 T315. From your data table, maybe the selected thermistor is not correct, the diferences are important at high temperatures.

BogusF, 2022-07-17:
I did it with the precompiled version, Ender3S1-F1-UBL-20220716.bin, to be exact. My own compilation wouldn't run.

mriscoc, 2022-07-17:
Please try with this T13 versions: 
[Experimental-Ender3S1-F1-UBL-T13-20220717.zip](https://github.com/mriscoc/Ender3V2S1/files/9128329/Experimental-Ender3S1-F1-UBL-T13-20220717.zip)
[Experimental-Ender3S1-F4-T13-20220717.zip](https://github.com/mriscoc/Ender3V2S1/files/9128334/Experimental-Ender3S1-F4-T13-20220717.zip)

BogusF, 2022-07-18:
gladly, here are my measurements:

```
set    measured
(300 | 293)
290 | 287
280 | 277
270 | 267
260 | 257
250 | 247
240 | 237
230 | 226
220 | 216
210 | 206
200 | 196
190 | 186
180 | 176
170 | 166
160 | 156
150 | 146
140 | 137
120 | 116
100 | 97

```
I checked the installation of the probe with the previous firmware and got the same temp measurements @ 150°, 205° and 250°C, so today's measurement should be comparable to yesterdays. I did in in steps of 10K because it really takes a lot of time waiting for the temp to settle, I hope that's okay. 
That last measurement @300°C got f-ed up sadly, the rockwool got smelly/burnt and I touched the setup which resulted in considerably bigger temp offset. 
Interesting how this table has a pretty constant offset of 3-4K.

BogusF, 2022-07-18:
>So, can we say with confidence that T13 match better the thermistor in the Sprite extruder Pro?

Sure looks like it to me. 8-)

I don't think (all of) the offset is thermal resistance, if that was the case, the offset would have to grow with increasing temperatures, since thermal resistance shouldn't change, but thermal loss through radiation, convection and heat conduction through the thermocouple wire would increase overall heat dissipation of the measurement setup. However, the opposite is true, from 240°C on up, the offset is only 3 degrees.

On the other hand, that could have a number of causes we also can't rule out with this setup. Doesn't matter though, it's much closer than before and print settings are manually adapted for specific printing needs anyways, so pinpoint precision isn't strictly neccessary. 
From my point of view this issue has been resolved a long time ago, I was already happy with the full range and bad precision, this is a big improvement.

closed 2022-07-18 by mriscoc

theSt33v, 2022-07-22:
> I'm releasing here: https://github.com/mriscoc/Special_Configurations/releases/tag/AM3950T13 the precompiled version to use the Sprite extruder Pro.

Where is the extended thermistor table for 13 that was used to compile this? Did you use the table that was suggested by BogusF?

mriscoc, 2022-08-02:
> > I'm releasing here: https://github.com/mriscoc/Special_Configurations/releases/tag/AM3950T13 the precompiled version to use the Sprite extruder Pro.
> 
> Where is the extended thermistor table for 13 that was used to compile this? Did you use the table that was suggested by BogusF?

I use this as starting point: https://www.thingiverse.com/thing:103668

theSt33v, 2022-08-02:
Do you not intend to release the extended thermistor table that you generated using this method?

mriscoc, 2022-08-02:
> Do you not intend to release the extended thermistor table that you generated using this method?
I'm not sure, I need more tests to make it official and also, I'm now researching an alternative method to take advantage of the 32bit boards.

Liep54, 2022-08-25:
> EDIT: the precompiled firmware for T13 has an extended table to be able to reach higher temperatures.
> 
> The absolute maximum is defined by the highest value in the calibration table for a given thermistor, the safe maximum is that maximum value minus 15 °C of overshoot, for a stock Marlin table: (300-15) = 285 °C

Where can I find the recompiled firmware

Liep54, 2022-08-26:
Thank you for your response Migual, but this is Chinese to me, as I don't
have a clue what to do. I am very new at this, and don't understand gcode
at all.
Then my laser module is not working with this firmware.

Regards

Lewis Liebenberg

mriscoc, 2022-08-26:
Look at the end of the page
[image]

Liep54, 2022-08-26:
Hi

I have the Ender 3 s1 with the F1 board, and I don't want the UBL.
Will future updates be able to use the laser.

Regards

Lewis Liebenberg

Liep54, 2022-08-29:
Hi Migual

Will future updates of the fantastic professional firmware be able to use
the Creality laser module?

Regards

Lewis Liebenberg

mriscoc, 2022-08-29:
I don't have access to a laser module and don't have experience with it, please use or open other issue because this is about a different topic.