mriscoc/Ender3V2S1 #219: Ender 3 v2 4.2.2 neopixel support?

G1t1t, 2022-07-12:
Your link to Ender 3 v2 4.2.2 neopixel support is missing. I’ve tried compiling your firmware with know settings to enable neopixels and menu nothing is working? 

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

mriscoc, 2022-07-12:
https://github.com/mriscoc/Special_Configurations/

mriscoc, 2022-07-16:
Sources updated

closed 2022-07-16 by mriscoc