mriscoc/Ender3V2S1 #220: Printer Completely dies

VaioHaZZard, 2022-07-13:
**Describe the bug**
when trying to cancel a print mid print the printer goes  black (dies) as soon as you click accept.  Have to cycle power switch to get printer back.  

**To Reproduce**
Steps to reproduce the behavior:
1. Go to  Cancel Print
2. Click on  Accept

**Version (please complete the following information):**
Ender3V2-427-BLTUBL-20220527

**Additional context**
This has happened on 3 out of 4 canceled prints.  
Fresh formatted memory card (card is not stock printer card) 
Have not had this issue with my 4.2.2 boards.

Again Miguel,  you are amazing,  keep up the good work.

mriscoc, 2022-07-16:
Because I must to do several test, I cancel demo many prints and never got that behaviour. There is also no similar report.

VaioHaZZard, 2022-07-16:
no worries.  I will see if the issue comes up on your new firmware release.  Looking forward to using it.

VaioHaZZard, 2022-07-22:
Was able to capture the printer dying.  Had to switch back to firmware [Ender3V2-427BLT-UBL-20220419.bin] due to poor print quality on the current release (will recompile with stepper smoothing and try again)  Now the printer dies after the purge in the start G-code.   Tried everything to try to narrow down where the issue might be (memory card, slicer, power strip etc..)  Was going to recompile latest firmware with stepper smoothing and test.   Video link below to show issue.  Happens every attempt. 
https://photos.app.goo.gl/bGKxT8yjkLXFd54x7

mriscoc, 2022-07-22:
Where is connected your hotend fan? In stock configuration, the hotend fan (not the part cooling fan) is connected directly to power supply, so is not under control of the board. In your video I see that the hotend fan stops working, if it is stock connected, then you have a severe power supply failure.

VaioHaZZard, 2022-07-22:
hotend fan is connected to power supply.  Everything is working as normal and then all shuts off at the same time. Have to cycle the power switch to re-boot.  The power supply is the only component I have not tested because I do not have a spare.  I will open the unit and check for loose connections.   thank you

VaioHaZZard, 2022-07-22:
Sorry,  I misspoke (was thinking of my other printer).   This is an Ender 2 Pro with a 4.2.7 board and Ender 3 V2 display.  This power supply only sends out 3 wires to the main board.  Pos Neg and Ground.  Hotend fan is then fed from the power header on the main board.  All connections are tight with no apparent visual issues inside.

VaioHaZZard, 2022-07-22:
Have gone back to firmware [Ender3V2-427-BLT-20220716.bin] and benchy started with no issues.  Will monitor and see if shutting down issue comes back

mriscoc, 2022-07-22:
> 

Eventually that can happen again, maybe the power supply doesn't support the stepper speed and acelleration increased in the latest releases, you can decrease them using the gcodes:
```gcode
M203 X300.00 Y300.00 Z5.00 E25.00 ;Setup machine max feedrate
M201 X500.00 Y500.00 Z100.00 E1000.00 ;Setup machine max acceleration
```

VaioHaZZard, 2022-07-22:
That was it.  Guess I better look for a Meanwell.  Thank you for your help

closed 2022-08-17 by mriscoc