mriscoc/Ender3V2S1 #224: Creality Ender3V2 Professional Firmware 20220527

gomitologiallo, 2022-07-16:
Greetings
I have installed, after making an UPGRADE on my Creality Ender3V2 (Creality Silent Mainboard 4.2.7 / Creality Sprite Pro Extruder Kit / Cr Touch), the Professional Firmware 202205527 Version 4.2.7 BLT.
The problem that now Z offset varies every time I turn the printer off and on.
Before loading this Firmware everything worked fine but now even if I reinstall the previous Firmware, nothing changes regarding Z Offset.
Is this problem known?
Can you help me please?
Regards

mriscoc, 2022-07-16:
Some CRTouch doesn´t work consistently. But if with your previous firmware it worked, I recommend you to return to that version. Flashing the firmware is an reversible operation and nothing but configuration is left in the printer, So, if you change the firmware and load the default settings you always we return to the previous status.

gomitologiallo, 2022-07-16:
Hello
I tried to go back to the previous version but the Z Offset also varies with the previous version (and before downloading the new firmware it worked fine)

mriscoc, 2022-07-16:
If your CRTouch is also failing with the old firmware you can have a hardware problem with it, check your wires and conections.

Moobien, 2022-07-17:
I can attest this has been happening for me over the last few firmware versions. Trammed  bed, set zoffset with 0.1mm feeler.  go to print and after the g29 has run the nozzle is a lot higher than 0.1mm

gomitologiallo, 2022-07-17:
> If your CRTouch is also failing with the old firmware you can have a hardware problem with it, check your wires and conections.

The CR touch works fine.
I think the problem is the firmwareZ offset varies after each print and each time the printer is turned on.
And if you read Moobien Green's latest comment that I'm not the only one who has this problem.

Moobien, 2022-07-17:
> The CR touch works fine. I think the problem is the firmwareZ offset varies after each print and each time the printer is turned on. And if you read Moobien Green's latest comment that I'm not the [...]

Yeah my cr touch is fine according to the M48 diagnostic

mriscoc, 2022-07-17:
If your hardware is not working with a version where it previously worked with the same configuration then it is a hardware problem.

closed 2022-07-17 by mriscoc

Moobien, 2022-07-17:
Only time i never seemed to have this issue was on stock firmware