mriscoc/Ender3V2S1 #225: Ender3 V2/S1 Professional Firmware 20220716 won't load on latest Ender 3V2 with DACAI Screen

GT-Geo, 2022-07-16:
I have 7 Ender 3 V2's, I was able to install this update on 5 of them but the newest ones won't load it. They all have 4.2.2 main boards. The only difference with the new ones is they have DACAI screens.

mriscoc, 2022-07-17:
Try renaming the firmware file and flash again, also make sure that you have the stock firmware in your screen.

GT-Geo, 2022-07-17:
I reverted back to stock for both board and screen. It flashed this time. Thanks for your help, I'll close this now since it's not really an issue.

closed 2022-07-17 by GT-Geo

titogarrido, 2022-07-22:
Hi! I have just bought an S1 and I am not able to flash the latest version. It is using stock firmware 3.0.3 (so I guess it is an F4), I have renamed the file to something like EnderS1.bin but still doesn't work. I have formatted the SD card and put just the bin file there but nothing happens. Any clue?