mriscoc/Ender3V2S1 #227: False runout sensor triggering

Moobien, 2022-07-17:
**Describe the bug**
A clear and concise description of what the bug is.

Updated to the Ender3V2-422-BLT-20220716 firmware, did my usual full calibration, during my prints, EVERY time my print head returned to the front left corner the screen would say Runout Sensor and then do the procedure for changing filament.  tried several prints and it always happens.  rolled back to previous firmware, recalibrated and prints fine without this issue.

**Expected behavior**
A clear and concise description of what you expected to happen.

honestly expecting it to not falsly trigger a run out sensor activation.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version (please complete the following information):**
Firmware version and release date

Ender3V2-422-BLT-20220716

**Additional context**
Add any other context about the problem here.

Moobien, 2022-07-17:
EDIT: using the creality run out sensory on my machine.

VaioHaZZard, 2022-07-17:
Also had this issue with 427-UBL-20220716.  would trigger when "card init error" came up.  cycled printer and used different memory card and seemed to print ok.

mriscoc, 2022-07-17:
https://github.com/mriscoc/Ender3V2S1/issues/230#issuecomment-1186571663

highground88, 2022-07-18:
Same issue - running via Octoprint, original E3 S1 sensor

Lungenpest, 2022-07-21:
Same here - running also via Octoprint. Ender3 S1, original sensor. **Advanced** / **Filament Settings** the active mode  from motion to **HIGH** worked for me. Thanks mriscoc for the help. You are the best!!!

MarkforU, 2022-07-23:
I'm kind of running into same issue here. Got the latest Build if the Ender 3 V2 for 4.2.2 board, I Must say, Best Firmware by Far that I've seen!!! Have the Creality sensor runout switch that runs by way of a toggle switch technically. 
Issue, whether running from Octoprint or SD card in machine any print. Print starts, draws line's, then goes and draws the skirt, starts on first layer of print, then when it goes to layer change, filament sensor unloads backwards a 100mm of filament out of the hotend after going to back right hand corner of printer, asks if you want to purge or continue after it draws back and loads it back up. I've tried Motion which I read in a post, that didn't work, found the help page here and tried high, still did the same thing. I mean printing is fine except for every level it list a runout on display and aborts printing feeds filament backwards, which I found out I could adjust down to 25mm, goes to back right corner, you hit button and it reloads and shoots out a gloob.Then asks if you want to do it again or continue to print, you hit continue after cleaning the head, and it goes right back to next layer does it and on z lift for next layer goes back to back righthand corner. 
This is happening with 2 different printers with the 4.2.2 board and UBL firmware. BTW, that is suck a nice feature. Every bottom layer is perfect!!!

mriscoc, 2022-07-23:
> I'm kind of running into same issue here. Got the latest Build if the Ender 3 V2 for 4.2.2 board, I Must say, Best Firmware by Far that I've seen!!! Have the Creality sensor runout switch that runs by way of a toggle switch technically...

If you have a normal runout sensor active mode must be HIGH or LOW. MOTION is only for BTT Smart sensor.

closed 2022-07-24 by mriscoc