mriscoc/Ender3V2S1 #229: 20220527 4.2.2 512K singing on diagonal movement

SorenRR, 2022-07-17:
**Describe the bug**
Board: 4.2.2 STM32F103 RET6 with no mods.
Printer is "singing" on diagonal movement.

**To Reproduce**
Steps to reproduce the behavior:
1. Load Ender3V2-422-MM-20220716.bin firmware
2. Load default config + use attached config.
3. Print "something" --> Printer is "singing"
4. Load Ender3V2-422-MM-20220527.bin firmware
5. Load default config + use attached config.
6. Print same "something" --> Printer is "quiet"

Configuration:
Sound off ;-)
Bed; 230x230
Xmin Xmax; 4, 244
Ymin Ymax; 3, 288
Zmax; 250
X; 80.4 steps/mm
Y; 80.5 steps/mm
Z; 402.4 steps/mm
E; 97.2 steps/mm

**Expected behavior**
Printer to remain "quiet"

**Version (please complete the following information):**
20220716 - MID-YEAR release Ender3V2-422-MM-20220716.bin

VaioHaZZard, 2022-07-17:
Also have this issue on Ender 3 S1 and Ender 2 Pro with 4.2.7 board.   Noticed it immediately when homing the printers with 20220716 firmware.  This was not present on all previous versions of firmware.  Not sure if it's an actual issue, but it's definitely different.

mriscoc, 2022-07-17:
With the latest version ADAPTIVE_STEP_SMOOTHING was disabled for increase movement precision, that increase the sound if the steppers move to high speed because that configuration limit the speed.
To return to a more silent operation you can limit the printer speed to the original stock values with:

`Control / Motion`
Speed menu:   X=300, Y=300, Z=5, E=25
Acceleration menu: X=500, Y=500, Z=100, E=1000

Or using the G-codes:
```gcode
M203 X300.00 Y300.00 Z5.00 E25.00 ;Setup machine max feedrate
M201 X500.00 Y500.00 Z100.00 E1000.00 ;Setup machine max acceleration
```

closed 2022-07-17 by mriscoc

jxer, 2022-07-18:
there is a reason why people use pre compiled firmware, your not even open to making this an option? The printer is easily 3 times louder then before.

jxer, 2022-07-18:
> ADAPTIVE_STEP_SMOOTHING can be re-enabled in configuration_adv.h
> 
> [MarlinFirmware/Marlin#23958](https://github.com/MarlinFirmware/Marlin/issues/23958)

thats for a ski board. I am running a 4.2.2 creality board.

SorenRR, 2022-07-19:
Custom M201/M203 settings do not really change anything.

I have my printer (with super quiet fans) next to my workstation, so I ended up compiling a version with ADAPTIVE_STEP_SMOOTHING enabled.

Problem solved!

jxer, 2022-07-19:
> Custom M201/M203 settings do not really change anything.
> 
> I have my printer (with super quiet fans) next to my workstation, so I ended up compiling a version with ADAPTIVE_STEP_SMOOTHING enabled.
> 
> Problem solved!

any chance its the creality 4.2.2 bltubl version? if so can you provide a download link?

SorenRR, 2022-07-20:
> Just setting the federate back to stock got it quiet again. Don't think its a big deal guys.

Not for me, Comparing the default values of M201 and M203 for this firmware and the previous firmware show that only E-steps Max has been raised from 25 to 60.
Using the extreme settings speed(1000,1000,40,200) and acceleration(3000,3000,300,9999) with "ASS" enabled in this new firmware and the printer is dead quiet! Printing ASA at 200 mm/s 0.3 mm layer hight with a 0.6 mm nozzle and the finished product is useable but looks like sh**.

cvail55, 2022-07-20:
> Not for me, Comparing the default values of M201 and M203 for this firmware and the previous firmware show that only E-steps Max has been raised from 25 to 60. Using the extreme settings [...]

I like that solution much better. It does seem to be louder than it was stock after hearing it running today, it helped but its still order than it originally was. I guess ill roll back to previous firmware. Kinda silly, I don't see why this was changed. NOBODY here wants there printer sounding like a dot matrix printer lol

cvail55, 2022-07-21:
> > Kinda silly, I don't see why this was changed. NOBODY here wants there printer sounding like a dot matrix printer lol
> 
> Because there is some bug with UBL and SKR boards, why he disabled it for all boards is beyond me. Also the print quality (yes i suffered through one print) had noticeably more artifacts i assume from the shaking. I also reenabled ASS, and fixed some other stuff, and its back to normal.

I don't know how to compile my own yet. Guess I need to learn.

arthurkok2, 2022-07-21:
> Because there is some bug with UBL and SKR boards, why he disabled it for all boards is beyond me. Also the print quality (yes i suffered through one print) had noticeably more artifacts i assume [...]

do you have these changes committed to your own fork? would love to see them. thanks!

jxer, 2022-07-22:
> do you have these changes committed to your own fork? would love to see them. thanks!

Sorry no, I have 0 interest in maintain a fork. I was annoyed to begin with that I had to learn this, but he pretty much has refused every single suggestion made thus far.

TheHolyRoger, 2022-07-23:
*sigh* guess I'll maintain my own fork now

arthurkok2, 2022-07-23:
> 

i made my fork here for ender 3 S1 https://github.com/arthurkok2/Ender3V2S1/tree/arthur-s1-changes

TheHolyRoger, 2022-07-26:
Looks like @mriscoc is in the process of re-enabling ADAPTIVE_STEP_SMOOTHING looking at the config files so thanks for listening man!

lackY0704, 2022-07-29:
> Mit der neuesten Version wurde ADAPTIVE_STEP_SMOOTHING für eine bessere Bewegungsgenauigkeit deaktiviert, die den Klang erhöht, wenn sich die Stepper auf hohe Geschwindigkeit bewegen, da diese Konfiguration die Geschwindigkeit begrenzt. Um zu einem lautloseren Betrieb zurückzukehren, können Sie die Druckergeschwindigkeit auf die ursprünglichen Bestandswerte beschränken mit:
> 
> `Control / Motion` Geschwindigkeitsmenü: X = 300, Y = 300, Z = 5, E = 25 Beschleunigungsmenü: X = 500, Y = 500, Z = 100, E = 1000
> 
> Oder mit den G-Codes:
> 
> ```gcode
> M203 X300.00 Y300.00 Z5.00 E25.00 ;Setup machine max feedrate
> M201 X500.00 Y500.00 Z100.00 E1000.00 ;Setup machine max acceleration
> ```
Speed ​​menu: X=300, Y=300, Z=5, E=25
Acceleration menu: X=500, Y=500, Z=100, E=100
I've tried everything no change
Ender 3v2 board 4.2.7. and board 4.2.2. This firmware is silent
Ender3V2-427-BLT-20220527.bin. :
All after that are loud especially the X axis. Manufacturers try to make printers quieter and the firmware makes the printer loud again.

duettra, 2022-07-30:
> > 
> 
> i made my changes here for ender 3 S1 https://github.com/arthurkok2/Ender3V2S1/tree/arthur-s1-changes

@arthurkok2 
Can you add the binary of Ender3S1-F4 firmware Ender3S1-F4-20220716.bin to your fork, please?

No interest to get into compiling the firmware from sources; I'd rather swap to Klipper in this case.

Many thanks!

arthurkok2, 2022-07-30:
> > > 
> > 
> > 
> > i made my changes here for ender 3 S1 https://github.com/arthurkok2/Ender3V2S1/tree/arthur-s1-changes
> 
> @arthurkok2 Can you add the modified binary of Ender3S1-F4 firmware with enabled ADAPTIVE_STEP_SMOOTHING to your fork, please?
> 
> No interest to get into compiling the firmware from sources; I'd rather swap to Klipper in this case.
> 
> Many thanks!

try this [firmware-20220730-121106.zip](https://github.com/mriscoc/Ender3V2S1/files/9226870/firmware-20220730-121106.zip)

duettra, 2022-07-30:
> try this [firmware-20220730-121106.zip](https://github.com/mriscoc/Ender3V2S1/files/9226870/firmware-20220730-121106.zip)

OMG finally my E3S1 F4 is silent again! Done my usual calibration and all runs flawlessly

Many thanks @arthurkok2 !

mriscoc, 2023-03-04:
Cleaned up thread for use as reference.