mriscoc/Ender3V2S1 #230: Filament Change before starting?

ccoupe, 2022-07-17:
I tried  Ender3V2-422-BLT-20220716 and something very strange happens when starting a print. After the purge line when the head should move to the beginning point for the skirt it moves to the back right and unloads the the filament - it's doing a change filament! With the pause and reload.  This makes it very unusable.   If it helps, I've got a Crtouch, not a bltouch.

mriscoc, 2022-07-17:
If you don't have a filament runout sensor please disable it in `Advanced/Filament Settings` if you have a sensor, then change the active mode to `HIGH/LOW`

mriscoc, 2022-07-17:
Duplicate of https://github.com/mriscoc/Ender3V2S1/issues/227

closed 2022-07-17 by mriscoc

ccoupe, 2022-07-17:
That was it! Thank you. We do like simple solutions.