mriscoc/Ender3V2S1 #231: steppers making louder noises on latest firmware

jxer, 2022-07-18:
Universal 20220716 422bltUBL

e3v2,

I was about 5 firmwares back, and my printer made the normal noises (own 2) one is on jyers one is on this

after replacing the nozzle and upgrading to this code and a PID tune. I am now noticing significantly more noises coming out of my steppers, which seems very odd to me as nothing was touched at the stepper or voltage level

was there anything done in the firmware that could louder noises in the steppers? seems to be x/y and not z

jxer, 2022-07-18:
I am reading there is some kind of silent mode for the steppers? is this true?

mriscoc, 2022-07-18:
Duplicated of https://github.com/mriscoc/Ender3V2S1/issues/229

closed 2022-07-18 by mriscoc

mriscoc, 2022-07-18:
Currently there is no plan to be integrated as menu option, but can be re-enabled in the configuration_adv.h file for devices with 512K of program memory (RET6) we are reaching the program memory limit for the RCT6 devices.

jxer, 2022-07-18:
That issue is specific to skr boards per the marlin discussion page, why are you implementing it across creality boards?