mriscoc/Ender3V2S1 #232: SD card files sorting on the printer screen

TioSam13, 2022-07-19:
First of all, thanks for the wonderfull work on our firmware, it's a game changer with more options than creality stock for sure.

I would like to request a extra option on the configuration menu to choose between alphabetic and last modified files on the printer screen. Currently, we are locked with alphabetic order wich is ok, but lot's of times we have Stl's with similar file names

Add an option on the advanced menu to choose between alphabetical and last modified files.
Prusa already got this option compiled on their firmware as seen in the 2 images bellow

[image]
[image]

GeorgelT, 2023-02-15:
I'd just like to add myself also to this request. Wanted to submit an issue exactly with this request but saw this one exists already. Really annoying sometimes, I'm not sure how it sorts files right now, but I don't think it's alphabetical either(on 22 dec fw release on S1). 

I usually have my naming as nozzel size, time, model name, material, so all my files start like this: 0.4_20h_modelname_pla , but sometimes newest ones are first, sometimes last and they are definitely not alphabetical. Started a print already but will add a screenshot once it finishes.

mriscoc, 2023-02-15:
The filelist is sorted alphabetically if the number of files is less than 40.
You can configure the file sorting with the command [M34](https://marlinfw.org/docs/gcode/M034.html), use it in a [configuration file](https://github.com/mriscoc/Ender3V2S1/wiki/Configuration-files).

TioSam13, 2023-02-16:
> The filelist is sorted alphabetically if the number of files is less than 40, use SDSORT_LIMIT for increase the number of files sorted. You can configure the file sorting with the command [M34](https://marlinfw.org/docs/gcode/M034.html), use it in a [configuration file](https://github.com/mriscoc/Ender3V2S1/wiki/Configuration-files).

After following your instructions on the [configuration file](https://github.com/mriscoc/Ender3V2S1/wiki/Configuration-files) I coded the file like this and saved the file as: auto0.g

; Professional Firmware Configuration File
;=====================================================
C10                    ; Mark as a configuration file
M117 Applying configuration
;-----------------------------------------------------
M34 F1 S<bool>         ; SDCard Sorting
;-----------------------------------------------------
G4 S1                  ; Wait a second
M300 P200              ; Beep
M117 Configuration Applied
;-----------------------------------------------------

Now I have the SD CARD files in alphabetical instead of no order, wich is a progress, but is it possible to put the files in the Last Modified order?

sarah1331sarah, 2023-04-30:
I would still like this

TioSam13, 2023-05-01:
> I would still like this to sort in latest modified

Me too, is the best sorting option!

closed 2023-07-12 by github-actions[bot]