mriscoc/Ender3V2S1 #241: Bugs in file listing

hiltonjanfield, 2022-07-23:
In the Print file listing, there is a strange bug using the latest firmware (20220716). Two bugs, technically.

The file list shows a directory entry for `..` (previous directory) which I do not recall seeing in the previous firmware version. This is separate from the normal `Back` entry at the top of the list.

When moving the cursor up from the `..` entry to the `Back` entry, the title of the entry changes to the DOS filename of the first file in the list. For example, if the first file in list is `cube_24m_0.20mm_...`, the (..) entry will change to `CUBE_2~1`. This reverts when moving the cursor down from this entry to the first file in the list.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Print'
2. Go into any subdirectory
3. See '..' entry
4. Move cursor up and down past this entry.

**Expected behavior**
The unnecessary entry appears, and its title changes as described above.

[**Video**](https://drive.google.com/file/d/1TL9slTCr80NdGuU4qXFFl0rG_a1JY9dx/view?usp=sharing)

**Version (please complete the following information):**
Ender3S1-F1-UBL-20220716

mriscoc, 2022-07-25:
I received this report before, but I was unable to reproduce it. Maybe you could try to minimize the amount of files in the memory card where this error occurs, then zip the content and post it here.

hiltonjanfield, 2022-07-25:
With a little testing, I found that it only occurs with excessively long filenames, such as those output by PrusaSlicer.
The attached ZIP contains a single directory and a single tiny file. Having just this on a freshly FAT32-formatted SD card will cause the bug for me, when viewing the subdirectory
[calibration.zip](https://github.com/mriscoc/Ender3V2S1/files/9178160/calibration.zip).

mriscoc, 2022-07-26:
Thank you, I will test it with your file.

hiltonjanfield, 2022-07-26:
I noticed last night that this was happening in the root folder as well. I
had two files with very long PrusaSlicer-style names, and similar to the
reported issue, the first file (which on the display is in the same
position as `..` would be) would change names to match the second. This
time was different, however, in that the entire long filename was used, not
the DOS filename.

mriscoc, 2022-08-05:
I rewrote the select file menu, the next version will have this issue fixed.

hiltonjanfield, 2022-08-05:
I'm glad my information was able to help you sort it out. I might have looked into it myself but I am completely unfamiliar with firmware sources, and was never great with C++. Thank you for all your work in making these machines better! I hope Creality helps you out in some way.

mriscoc, 2022-08-05:
Thanks, I'm not affiliated o related to Creality in any way. It is only the brand of my printer.

mriscoc, 2022-08-17:
Fixed in the latest version.

closed 2022-08-17 by mriscoc

hiltonjanfield, 2022-10-11:
I know, but for all this work improving things they should take notice.
First advice to new users with problems on the Ender 3 Discord is usually
either install MRiscoC or set up Klipper, and one of those is much easier
than the other.