mriscoc/Ender3V2S1 #242: 1047 : PT1000 w/ 4.7k Ohms (E3D) max temp error on boot

mendozabrah, 2022-07-23:
E3D PT1000 thermistor plugged directly into Ender 3 v2 4.2.2 RCT6 board, produces MAX TEMP ERR after switching on printer. Firmware was compiled with `TEMP_SENSOR_0 1047`  

Thermistor functions correctly on other firmware (Jyer's and Unified) with `TEMP_SENSOR_0 1047`.

**To Reproduce**
1. Plug E3D RTD PT1000 temp sensor to 4.2.2 (RCT6) board Ender 3 V2 (no amplifer)
2. Change `TEMP_SENSOR_0 1` to `TEMP_SENSOR_0 1047` in `Configuration.h`
3. Update firmware
4. Turn on printer

**Expected behavior**
After printer boots up, the following errors show in the screenshots

**Screenshots**
[image]
[image]

**Version**
Ender3V2-422-BLT 20220716 Mid year release

ZipperZ, 2022-07-23:
Same here with 1010 sensor (PT1000, 1k pull-up), 427 board.

Error is triggered by 
(pos && r > temp_range[e].raw_max) 
in line 2353 of file "temperature.cpp"

Was not able to track it down even further, but still going deeper.

limamg, 2022-07-23:
This problem happened to me a couple of times when I was already printing, after I turned it off and on again it started working normally.
I noticed that when this message happens, the temperature of the extruder is oscillating.

ZipperZ, 2022-07-23:
Ok, I am not yet sure how that works on a standard Marlin, but basically  _TEMP_MAX_E(NR) macro functions incorrectly, because temp_range[NR].raw_max is not initialized. 

Sorry I am no way near "programmer"...
Very dirty workaround(temperature.cpp line ~2651):

   #if _MINMAX_TEST(0, MAX)
      _TEMP_MAX_E(0);
    #endif

Add line:

   #if _MINMAX_TEST(0, MAX)
      temp_range[0].raw_max = 65535;
      _TEMP_MAX_E(0);
    #endif

ZipperZ, 2022-07-23:
> Ok, I am not yet sure how that works on a standard Marlin, but basically _TEMP_MAX_E(NR) macro functions incorrectly, because temp_range[NR].raw_max is not initialized.
>
> Sorry I am no way near [...]

And now after "dirty workaround" M501 triggers MAXTEMP error...

mriscoc, 2022-07-24:
The thermistor table for the type 1047 is ordered in inverted form with respect to the type 1, that causes some compilation issues with the algorithm of the custom G-code C104.
Currently the firmware do not support this inverted table. I will look for a solution in a future version.

mriscoc, 2022-07-24:
Please test with the sources from: https://github.com/mriscoc/Ender3V2S1

ZipperZ, 2022-07-24:
With sensor 1010 the printer acts the same. 
Right after boot error is triggered.

If I initialize temp_range[0].raw_max, then I am able to boot and use the printer until command M501, which also triggers the error.

mendozabrah, 2022-07-24:
Likewise, the same with sensor 1047. Still throwing MAX TEMP error after boot

mriscoc, 2022-07-24:
I don't have that sensors to test, I will try to find what's triggering that error message.

ZipperZ, 2022-07-25:
With those options disabled the code does not compile

`Marlin\src\lcd\e3v2\proui\dwin.cpp: In function 'float Tram(uint8_t)':
Marlin\src\lcd\e3v2\proui\dwin.cpp:2452:19: error: 'MESH_MIN_X' was not declared in this scope
 2452 |       LIMIT(xpos, MESH_MIN_X, MESH_MAX_X);
      |                   ^~~~~~~~~~
Marlin\src\lcd\e3v2\proui\dwin.cpp:2452:31: error: 'MESH_MAX_X' was not declared in this scope
 2452 |       LIMIT(xpos, MESH_MIN_X, MESH_MAX_X);
      |                               ^~~~~~~~~~
Marlin\src\lcd\e3v2\proui\dwin.cpp:2453:19: error: 'MESH_MIN_Y' was not declared in this scope
 2453 |       LIMIT(ypos, MESH_MIN_Y, MESH_MAX_Y);
      |                   ^~~~~~~~~~
Marlin\src\lcd\e3v2\proui\dwin.cpp:2453:31: error: 'MESH_MAX_Y' was not declared in this scope
 2453 |       LIMIT(ypos, MESH_MIN_Y, MESH_MAX_Y);
      |                               ^~~~~~~~~~
*** [.pio\build\STM32F103RE_creality\src\src\lcd\e3v2\proui\dwin.cpp.o] Error 1
In file included from Marlin\src\lcd\e3v2\proui\../../../inc/MarlinConfigPre.h:66,
                 from Marlin\src\lcd\e3v2\proui\endstop_diag.cpp:22:`

After changing `EITHER(MESH_BED_LEVELING, AUTO_BED_LEVELING_UBL)` to "ANY(MESH_BED_LEVELING, AUTO_BED_LEVELING_UBL, AUTO_BED_LEVELING_BILINEAR)"
In Configuration_adv.h in line ~2105 the code compiled.

Now everything seems fine. The board boots, M501 does not cause error either.

Is it memory constraint of STM32F103RC (I assume this uC is used in 427?) If it is, then would it help if I change it to RE? Only I do not know what kind of bootloader is used and where to get it.

mendozabrah, 2022-07-28:
> With those options disabled the code does not compile
>
> `Marlin\src\lcd\e3v2\proui\dwin.cpp: In function 'float Tram(uint8_t)': Marlin\src\lcd\e3v2\proui\dwin.cpp:2452:19: error: 'MESH_MIN_X' was not [...]

This worked for me too, 1047 thermistor works perfectly ! Appreciate the fix @mriscoc and @ZipperZ

mriscoc, 2022-10-16:
I want to address this bug disabling the custom G-code C104 for PT1000 thermistor. Anyone to test some experimental builds?

ZipperZ, 2022-10-16:
> I want to address this bug disabling the custom G-code C104 for PT1000 thermistor. Anyone to test some experimental builds?

Hi, I could test it. Do not promise very fast feedback, but I will

NiLMoXiE, 2022-10-16:
With the options above commented out and ZipperZ's change I still get a compile error (I am using the source from special releases E3V2-422-BLT-UBL):

Marlin\src\lcd\e3v2\proui\dwin.cpp: In function 'void Draw_PrintDone()':
Marlin\src\lcd\e3v2\proui\dwin.cpp:563:5: error: 'Preview_Show' was not declared in this scope
  563 |     Preview_Show();
      |     ^~~~~~~~~~~~
Compiling .pio\build\STM32F103RC_creality\src\src\libs\least_squares_fit.cpp.o
Marlin\src\lcd\e3v2\proui\dwin.cpp: In function 'void DWIN_Print_Started()':
Marlin\src\lcd\e3v2\proui\dwin.cpp:1542:24: error: 'Preview_Invalidate' was not declared in this scope
 1542 |   if (Host_Printing()) Preview_Invalidate();
      |                        ^~~~~~~~~~~~~~~~~~
*** [.pio\build\STM32F103RC_creality\src\src\lcd\e3v2\proui\dwin.cpp.o] Error 1

Commenting out Preview_Show and Preview_Invalidate allowed it to compile but I can't test yet as I am waiting on parts.

mriscoc, 2022-10-16:
A 422 BLTUBL version could be fine for all of you?

ZipperZ, 2022-10-16:
> A 422 BLTUBL version could be fine for all of you?

I am using 427 and my sensor is 1010 (changed resistor on the board). Also the thermistor table extended from the "original"
Maybe you could create a branch for testing? (we would customize according our setups?)

mriscoc, 2022-10-16:
OK, it is ready to test: https://github.com/mriscoc/Ender3V2S1/tree/Ender3V2S1-NO_C104

NiLMoXiE, 2022-10-16:
Not compiling for me - getting a 'file not found error' -

FileNotFoundError: [Errno 2] No such file or directory: 'Marlin/lib/proui/stm32f1/libproui_abl.a':
  File "C:\Users\daver\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 186:
    env.SConscript(item, exports="env")
  File "C:\Users\daver\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\daver\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "E:\Test Firmware\Ender3V2S1-Ender3V2S1-NO_C104\buildroot\share\scripts\proui.py", line 53:
    shutil.copy(libpath+arch+'libproui_abl.a', libpath+'libproui.a')
  File "C:\Users\daver\.platformio\python3\lib\shutil.py", line 427:
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "C:\Users\daver\.platformio\python3\lib\shutil.py", line 264:
    with open(src, 'rb') as fsrc:

NiLMoXiE, 2022-10-16:
Getting 'file not found error' so won't compile - see below

FileNotFoundError: [Errno 2] No such file or directory: 'Marlin/lib/proui/stm32f1/libproui_abl.a':
  File "C:\Users\daver\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 186:
    env.SConscript(item, exports="env")
  File "C:\Users\daver\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\daver\.platformio\packages\tool-scons\scons-local-4.4.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "E:\Test Firmware\Ender3V2S1-Ender3V2S1-NO_C104\buildroot\share\scripts\proui.py", line 53:
    shutil.copy(libpath+arch+'libproui_abl.a', libpath+'libproui.a')
  File "C:\Users\daver\.platformio\python3\lib\shutil.py", line 427:
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "C:\Users\daver\.platformio\python3\lib\shutil.py", line 264:
    with open(src, 'rb') as fsrc:

mriscoc, 2022-10-17:
Sorry, my mistake, rename `Marlin/lib/proui/STM32F1/libproui.a` to `libproui_ubl.a`

mriscoc, 2022-10-17:
> Can I copy from the release branch or is a different version required?
No, please don't mix files from the release branch.

oddmolecules, 2022-10-17:
that seems to have it working for me, running a pid test now. I was getting a FileNotFoundError: [Errno 2] No such file or directory: 'Marlin/lib/proui/stm32f1/libproui_mbl.a'
Its the manual mesh version, not the ubl, just had to swap libproui_ubl.a to mbl and then make the ubl changes that i want. 

Thank you for taking the time to fix this for us minority of pt1000 users!! definitely appreciated!! will get back after testing further if there are any problems.

mriscoc, 2022-10-17:
> that seems to have it working for me, running a pid test now. I was getting a FileNotFoundError: [Errno 2] No such file or directory: 'Marlin/lib/proui/stm32f1/libproui_mbl.a' It's the manual mesh [...]

Manual mesh needs another libproui library. This FIRST test is only to verify that the origin of the bug is in the C104 code. I will prepare others tests to get firmware that can be used by anyone without taking care about the thermistor type.

Thank you for helping me to narrow where the problem is.

mriscoc, 2022-10-19:
I think that this issue is now fixed in the main branch.

wwian, 2022-10-20:
Miguel, will you be making precompiled binary files with this fix?

mriscoc, 2022-10-21:
> Miguel, will you be making precompiled binary files with this fix?

Typically, special configurations such as PT1000 thermistor are compiled by the users themselves, but it is possible to request a sponsored compilation, see the readme at https://github.com/mriscoc/Special_Configurations

WHOAREYOUAAA, 2022-10-23:
Getting the same error, I updated the firmware with the newest version today. I only changed the "BED_MAXTEMP" in configuration.h from 110 to 120. It My printer worked great for 3 days now it fails every time unless I use a precompiled marlin version which is so odd. Weird thing is that the printer will work sometimes only if an SD card is inserted. My printer's hardware is completely stock other than this firmware.

mriscoc, 2022-10-23:
> Getting the same error, I updated the firmware with the newest version today. I only changed the "BED_MAXTEMP" in configuration.h from 110 to 120. It My printer worked great for 3 days now it fails [...]

What thermistor type does your printer have? 1010, 1047?

mriscoc, 2022-10-23:
> Getting the same error, I updated the firmware with the newest version today. I only changed the "BED_MAXTEMP" in configuration.h from 110 to 120. It My printer worked great for 3 days now it fails [...]

This page is for issues with the thermistor PT1000 (type 1047 or 1010)

kiu345, 2022-10-28:
> I think that this issue is now fixed in the main branch.

Compiled with ZIP Ender3V2S1-20221002, still getting the error. Tried workaround, nothing changed. Any idea?

`    #if _MINMAX_TEST(0, MIN)
      _TEMP_MIN_E(0);
    #endif
    #if _MINMAX_TEST(0, MAX)
      temp_range[0].raw_max = 65535;  // Workaround
      _TEMP_MAX_E(0);
    #endif
    #if _MINMAX_TEST(1, MIN)
      _TEMP_MIN_E(1);
    #endif
`

mriscoc, 2022-10-30:
> > I think that this issue is now fixed in the main branch.
> 
> Compiled with ZIP Ender3V2S1-20221002, still getting the error. ~Tried workaround, nothing changed~. Any idea?
> 
> _Update: Workaround worked, a "clean all, then build it" did it..._

Use the main branch.

mriscoc, 2022-12-05:
No more user feedback

closed 2022-12-05 by mriscoc