mriscoc/Ender3V2S1 #245: Start Stop Print

egonwinter, 2022-07-23:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

egonwinter, 2022-07-23:
When I print with Octopi, there is no solution to stop a print, only in Cura ore Octopi, is it possible to do a Butten in the print Menü to stop the print? Thanks for your answer

mriscoc, 2022-07-23:
> 

https://github.com/mriscoc/Ender3V2S1/wiki/Octoprint

closed 2022-07-23 by mriscoc

egonwinter, 2022-07-27:
Hi Miguel

You send me the Link, I check the Link, but I don´t know what to do, I´m not that specialist in programming I´m just
A user, please advice what to do to have a Break and a Print Stop button on the display, thanks for your help.

Best regards

Egon Winter

Wilhelm – Schweizer – Straße. 10
56564 Neuwied

Mob. +49 170 6868491
Tel. +49 2631 4003157

Mail. ***@***.***

https://github.com/mriscoc/Ender3V2S1/wiki/Octoprint

mriscoc, 2022-07-27:
> Hi Miguel You send me the Link, I check the Link, but I don´t know what to do, I´m not that specialist in programming I´m just A user, please advice what to do to have a Break and a Print Stop button on the display, thanks for your help. Best regards 

You only need to follow the guide for configure your Octoprint:  https://github.com/mriscoc/Ender3V2S1/wiki/Octoprint/_edit#octoprint-gcode-scripts