mriscoc/Ender3V2S1 #247: Rotary Encoder Main Menu Selection Goes Beyond Advanced

grilparto, 2022-07-25:
**Describe the bug**
When rotating the rotary encoder CW and highlight "Advanced" on the main menu, I'm able to continue rotating beyond "Advanced" in that rotations are still registered, but no visible main menu items are selected.

Oddly enough, it seems like settings of some sort are being selected off-screen.  The settings would be based on the menu options under "Prepare".  For example, if I turn the knob one more time CW past Advanced, and then depress the rotary encoder, it will initiate Homing.  One additional time and it appears to activate the tramming when depressing the rotary encoder.  5 turns CW beyond Advanced (and depressing the knob) seems to bring me to the Z Probe Wizard.

**Expected behavior**
The selection options should stop at highlighting "Advanced" and not continue if the user continues to rotate the rotary wheel CW.

**Version (please complete the following information):**
Ender3 V2/S1 Professional Firmware 20220716, however I compiled it for the BTT SKR Mini E3 V3.0 board.

**Additional context**
As of right now I only have this setup on a test bench with USB power and the LCD.  I've not connected any of the peripherals (bed, hot end, endstops, BL touch, etc).

mriscoc, 2022-07-25:
Beyond Advanced there are the items of the shortcut of toolbar.

miskig, 2022-07-25:
I am having the same issue. The shortcut toolbar is there, but not shown. Current version pulled from source. Flashed the DWIN set for the LCD to be sure. Ender 3 BLUBL 4.2.2.

Flashing a precompiled release makes to shortcut bar work properly.

mriscoc, 2022-07-25:
Use the provided configuration files at https://github.com/mriscoc/Special_Configurations **without changes**, test it, then start to make one change at time.

grilparto, 2022-07-26:
That'll be a little difficult given the setup.  I did a git pull of the default branch yesterday to try and adapt the firmware to my Voxelab Aquila running an SKR mini E3 3.0 with a BL Touch and direct drive setup.   Ender 3 V2 would be the closest hardware to the printer (aside from the SKR mini) and would use the same DWIN type display and pinout.  I ended up using the Ender3V2-422-BLT configuration and config_adv as a baseline, and compared against the Ender 3 configuration from BTT https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V3.0/Marlin.

Not many changes were necessary other than a couple of things specific to the SKR mini and the Aquila's specs.  For example, making sure the correct driver types are set, the print bed size, stepper motor directions, probe offset, serial port number, that sort of thing.  Things I'd have to change or else the printer simply won't function or move the steppers in the right direction.  I think the focus should be related to the DWIN or LCD areas of the config files but I didn't have any reason to touch those areas.

I did modify the platform.ini so that it'd compile with STM32G0B1RE_btt.   No real modifications needed from what Ican see beyond changing the default_envs to `default_envs = STM32G0B1RE_btt`

Curiously enough, I did have to sort out a compile error with the dwin.cpp.  I'm not sure why it throwing an error though.

`expected ')' before 'STRING_DISTRIBUTION_TIME'`
expected ')' before ';' token

Looking at those areas of the code, it seems like it should be correct, so I don't know what the problem is.  Line 995 is just

`  DWINUI::Draw_CenteredString(271, F(STRING_DISTRIBUTION_DATE " " STRING_DISTRIBUTION_TIME));`

I tried to write is slightly different to see if it resolved the issue (added + signs to concatenate) but that won't work because they're invalid operands.  Also get an error:

`'STRING_DISTRIBUTION_TIME' was not declared in this scope; did you mean 'STRING_DISTRIBUTION_DATE'?`

To resolve the problem, I simply changed line 995 to:

`  DWINUI::Draw_CenteredString(271, F(STRING_DISTRIBUTION_DATE));`

After that it compiled fine (aside from needing to comment out a couple of warnings imposed by the SKR mini related to the DWIN pinout and jumpers that need to be left off the board).

I did notice that the current branch has recent revisions since the Universal 20220716 MID-YEAR release mentioned here https://github.com/mriscoc/Ender3V2S1/releases .  Just to be thorough, I grabbed the source code zip from that release and went through the same steps.  Had the same compile error.  Oddly enough, the platform.ini in the root matched my modified version for the SKR mini.  I had the same compile issue mentioned above, and I did notice that there were some differences in the dwin.cpp file for the ProUI.  I haven't had a chance to flash it yet though.

Think my next steps will be to try flashing the latest stock DWIN_SET again from the release branch, then maybe try to flash the same config using the source files from 7/16, and flashing the DWIN_SET contained within that source release.

grilparto, 2022-07-26:
Just noticed an Ender3V2-SKRME3V3-BLT config in the special configurations that I hadn't seen before.  I'll give that a try without changes to see what happens.  It appears that it should mostly satisfy things (won't print correct, but it should boot I think).

grilparto, 2022-07-26:
Having fun over here!  Used the following special configuration as is:
https://github.com/mriscoc/Special_Configurations/tree/main/Ender3V2-SKRME3V3-BLT

Compiled just fine (even with the `DWINUI::Draw_CenteredString(271, F(STRING_DISTRIBUTION_DATE " " STRING_DISTRIBUTION_TIME));` line that was giving an error earlier).  Also seeing the shortcut bar.

I have zero idea what could've made an impact here but I'll be doing a compare on the four files against the special config.

grilparto, 2022-07-26:
Well, I still haven't found the config difference that's causing the shortcut bar to disappear, but I figured out what caused the compiler errors:

```
expected ')' before 'STRING_DISTRIBUTION_TIME'
expected ')' before ';' token
```

For some reason I'd commented out `#define CUSTOM_VERSION_FILE Version.h // Path from the root directory (no quotes)`

Once I uncommented that line, it compiled fine.  Can't recall if it was commented out in one of the config files I was building off of or if I'd commented it out making a comparison with the SKR mini config BTT made.

Sleep time for now, more digging tomorrow.

mriscoc, 2022-07-26:
Because the issue of rotary enconder was uncovered maybe it is better to close this issue and open another related to Aquila and your current progress.

grilparto, 2022-07-26:
Yeah, makes sense to close this as I'm guessing it's an issue related to a goof up in my config that I haven't found yet.  Probably a feature I enabled or disabled that isn't specific or essential to the aquila or SKR mini either. I can always just post my mistake here once I find it.  If there's anything else that's critical that needs to be configured to get this working on an aquila with the SKR mini e3, I can make a post or a guide.  But aside from the shortcut bar, seems like it's otherwise working fine.

closed 2022-07-26 by grilparto

grilparto, 2022-07-26:
Figured it out.  Posted new bug here that's specific to it.

https://github.com/mriscoc/Ender3V2S1/issues/249

classicrocker883, 2022-07-27:
for using an Aquila LCD, reverse the encoder direction when compiling.

classicrocker883, 2022-07-27:
> Use the provided configuration files at https://github.com/mriscoc/Special_Configurations **without changes**, test it, then start to make one change at time.

I am having the same issue where the toolbar is not showing.

the encoder goes past Advanced menu, and the toolbar menu options are "invisible".

I am able to select them, but their icons are simply not showing up.
I used the special configuration, and only the release .bin has it where it shows up. I used both the .zip main and from source on release.

grilparto, 2022-07-27:
In my case, it occurred because I changed the author in the config.h file since I was changing settings relevant to my printer.  The creator mentioned that this was by design.  See this separate issue.  https://github.com/mriscoc/Ender3V2S1/issues/249

Also, yep same experience with the rotary encoder wiring.  Just had to swap the A and B wires.

I ended up dropping this FW in favor of compiling Marlin from scratch now that they've baked in the ability to use Jyers UI within the config.h.  Having issues with my X axis losing its position when leveling but that's a separate issue of course.