mriscoc/Ender3V2S1 #248: Ender 3 S1 - Preview STL

pacocatech, 2022-07-25:
Hello,
I'm using the latest version of the FW that you released for ender3 s1 and I updated the display as per your post.
I did the entire flow of the python scripts to generate the preview of the file on the printer, however, when I put it to print my display always looks like this.

Is there anything I can do to resolve it?

[image]

mriscoc, 2022-07-25:
G-code preview is only compatible with DWIN screens not DACAI, check here: https://github.com/mriscoc/Ender3V2S1/wiki/How-to-generate-a-gcode-preview#dacai-screen-units-dont-support-g-code-preview

Compare your screen unit with these pictures: https://github.com/mriscoc/Ender3V2S1/wiki/How-to-update-the-display

limamg, 2022-07-26:
My Ender 3 V2 screen is DACAI and it shows the qrcode below the print information. I will post pictures of her.
[image]
[image]

closed 2022-07-26 by pacocatech