mriscoc/Ender3V2S1 #249: Changing STRING_CONFIG_H_AUTHOR in configuration.h breaks the DWIN shortcut toolbar

grilparto, 2022-07-26:
**Describe the bug**
When editing the Configuration.h file and recompiling the firmware for the BTT SKR mini E3 V3.0 (STM32G0B1RE_btt), editing the STRING_CONFIG_H_AUTHOR variable causes the shortcut toolbar to not appear on the LCD screen.

**To Reproduce**

1. Start with [Release 20220716](https://github.com/mriscoc/Ender3V2S1/tree/20220716) and [Special Configuration Ender3V2-SKRME3V3-BLT](https://github.com/mriscoc/Special_Configurations/tree/main/Ender3V2-SKRME3V3-BLT) as a baseline.
2. Open the Configuration.h file and change [STRING_CONFIG_H_AUTHOR](https://github.com/mriscoc/Special_Configurations/blob/ddaab6be0100bc6e5bc97e1cfebb9fad53b0c704/Ender3V2-SKRME3V3-BLT/Configuration.h#L66) as follows:

Original:
`#define STRING_CONFIG_H_AUTHOR "Miguel A. Risco-Castillo (MRiscoC)" // Who made the changes.`
Modified:
`#define STRING_CONFIG_H_AUTHOR "foo" // Who made the changes.`

3. Compile (Build) firmware using PlatformIO/VSCode STM32G0B1RE_btt
4. Flash firmware to BTT SKR mini E3 V3.0 via microSD card
5. Observe missing shortcut bar from UI on LCD after firmware is installed

**Expected behavior**
Changing the author of the Config.h file shouldn't unexpectedly impact the UI on the LCD.

**Version:**
- Downloaded source associated with https://github.com/mriscoc/Ender3V2S1/tree/20220716
- Replaced configuration.h, configuration_adv.h, platformio.ini, and Version.h files with those from [Special Configuration Ender3V2-SKRME3V3-BLT](https://github.com/mriscoc/Special_Configurations/tree/main/Ender3V2-SKRME3V3-BLT).

mriscoc, 2022-07-26:
Behaviour confirmed, change the AUTHOR of the firmware causes that the ProUI library doesn't work as expected. For use the ProUI you must to keep the AUTHOR name unchanged.

miskig, 2022-07-27:
Are there any other limitations in this firmware when changing the author of the configuration.h?

mriscoc, 2022-07-27:
> Are there any other limitations in this firmware when changing the author of the configuration.h?

If you want to change the firmware author, I recommend using the `DWIN_LCD_PROUI` flavor from the official Marlin repository or disable the ProUI extensions commenting out the lines in `configuration.h`:

```cpp
// Professional firmware features:
#define ProUIex 1
#define HAS_GCODE_PREVIEW 1
#define HAS_TOOLBAR 1
```

miskig, 2022-07-27:
Fair enough. 

Why did you implement this check?

mriscoc, 2022-10-04:
It is not a bug but a behavior by design.

closed 2022-10-04 by mriscoc