mriscoc/Ender3V2S1 #252: Correct print resume after filament change fails

RoP09, 2022-07-26:
**Describe the bug**
At filament change triggered manually or by runout sensor the printhead moves to back right position but moves a little too far (gives cracking sounds from the motor/drivetrain).

**To Reproduce**
Press filament change button while printing

**Expected behavior**
Filament change works without errors (actually it was working fine with previous FW version from May using the same settings).

**Version (please complete the following information):**
Latest FW from this page

**Additional context**
After print resume, the printer is not resuming on the same position, but is off by several centimeters (prints into empty space basically).

mriscoc, 2022-07-27:
Maybe your printer is not stock, check your parking position and X/Y max position: https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides

RoP09, 2022-07-27:
Hm, did a erase settings after FW Update as usual. But i will double check this again.

Definately was not an issue with your previous FW.

RoP09, 2022-07-27:
Checked settings:
X bed size 230
Y bed size 230
X max Pos 245
Y max Pos 231
Z max Pos 250

Park x 240
Park y 220
Park z 20

RoP09, 2022-07-28:
> Maybe your printer is not stock, check your parking position and X/Y max position: https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides

my fault indeed, had an smaller (rubber) right endstop since swtiching to linear rails! Can be closed, sorry for the inconvenicence!

mriscoc, 2022-07-29:
Closed by request

mriscoc, 2022-07-29:
Closed by request

closed 2022-07-29 by mriscoc