mriscoc/Ender3V2S1 #265: HS mode for CR touch recomendation may need updating

hayden-t, 2022-07-31:
In several wiki pages it is recommended to disable HS mode.

Ive found however that on my ender S1 pro F4 probe test or leveling fails without it.

thanks

mriscoc, 2022-08-01:
HS mode must be disabled for mesh leveling bigger than 5x5 points.

closed 2022-08-01 by mriscoc

hayden-t, 2022-08-01:
why you dismiss me ?? HS mode fails probe usage any size, with hs mode on i can do 9x9 mesh no problem

mriscoc, 2022-08-01:
> why you dismiss me ?? HS mode fails probe usage any size, with hs mode on i can do 9x9 mesh no problem

That behavior is very particular, I don't have any report about that. If non HS mode is failing for you that could be a indication of hardware failure and not a firmware failure. However, if the HS mode is working for you, that is better, but always check the resulting mesh leveling.

hayden-t, 2022-08-01:
HS mode works, the mesh matches the contour of my plate as witnessed by printing and using metal ruler

hayden-t, 2022-08-01:
https://github.com/mriscoc/Ender3V2S1/issues/195#issuecomment-1201521918