mriscoc/Ender3V2S1 #266: Media Init Fail message

G1t1t, 2022-07-31:
I am getting a Media init fail message, if I open the Print icon without sd card inserted or Ifuse a micro sd card extension. Media Init fail message will not clear without powering down the printer? 

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version (please complete the following information):**
Firmware version and release date

**Additional context**
Add any other context about the problem here.

cr08, 2022-07-31:
I'm seeing the same thing on my V2 (4.2.2 BLT). Doesn't seem to negatively affect anything, just a cosmetic glitch.

mriscoc, 2022-08-01:
Without a memory card the printer can't initialize it before of list the files inside it.

closed 2022-08-01 by mriscoc