mriscoc/Ender3V2S1 #269: [FR] (Fysetc Cheetah V2 support)

V1ntex, 2022-08-02:
### Is your feature request related to a problem? Please describe.

Hello, i'm trying port this firmware for cheetah v2, but i have some errors.

There are changes what i did:
### Configuration.h:
```
...
#define MOTHERBOARD BOARD_FYSETC_CHEETAH_V20
...
#define SERIAL_PORT -1
...
//#define PRINTCOUNTER 
...
```

### Conditionals_adv.h:
```
...
#ifndef LCD_SERIAL_PORT
  #if HAS_DWIN_E3V2 || IS_DWIN_MARLINUI || HAS_DGUS_LCD
    #if MB(BTT_SKR_MINI_E3_V1_0, BTT_SKR_MINI_E3_V1_2, BTT_SKR_MINI_E3_V2_0, BTT_SKR_MINI_E3_V3_0, BTT_SKR_E3_TURBO, FYSETC_CHEETAH_V20)
...

```

### stm32f4.ini:
```
...
build_flags        = ${stm32_variant.build_flags} 
                    -DMCU_STM32F401RC -DSTM32F401xC
                    -lproui -LMarlin/lib/proui                            
extra_scripts      = ${stm32_variant.extra_scripts}
                    pre:buildroot/share/scripts/proui.py
...
```

### platformio.ini:
```default_envs = FYSETC_CHEETAH_V20```

First power on is ok, but second power on is freeze main menu.

### Are you looking for hardware support?

Fysetc Cheetah V2

### Describe the feature you want

Support board

mriscoc, 2022-08-03:
Hi, If your first boot was successful then the firmware compile was right.
If the successive boot is unsuccessful, there is some default causing the freeze.
Try disabling in configuration.h file:
```cpp
// Professional firmware features:
#define ProUIex 1
#define HAS_GCODE_PREVIEW 1
#define HAS_TOOLBAR 1
```

V1ntex, 2022-08-05:
@mriscoc, Hi, Disabling didn't help =(
But if i choose incorrect eeprom (specially), then every starting is ok, but the setting are not savedon each restart and i get an error "eeprom version failed"

closed 2022-08-09 by V1ntex

mriscoc, 2022-08-09:
Were you able to found the problem with settings?