mriscoc/Ender3V2S1 #270: Make Titles Centered

classicrocker883, 2022-08-04:
In bold is the line to add to change to make the titles centered. I think this is an important change

file
proui --> **menus.cpp**

void Draw_Title(TitleClass* title) {
  DWIN_Draw_Rectangle(1, HMI_data.TitleBg_color, 0, 0, DWIN_WIDTH - 1, TITLE_HEIGHT - 1);
  if (title->frameid) 
    DWIN_Frame_AreaCopy(title->frameid, title->frame.left, title->frame.top, title->frame.right, title->frame.bottom, 14, (TITLE_HEIGHT - (title->frame.bottom - title->frame.top)) / 2 - 1);
  else
**DWINUI::Draw_CenteredString(false, DWIN_FONT_HEAD, HMI_data.TitleTxt_color, HMI_data.TitleBg_color, (TITLE_HEIGHT - DWINUI::fontHeight(DWIN_FONT_HEAD)) / 2 - 1, title->caption);** //changed here

}

mriscoc, 2022-08-04:
Thanks, but it's a matter of taste. I prefer uncentered menu titles.

classicrocker883, 2022-08-06:
> Thanks, but it's a matter of taste. I prefer uncentered menu titles.

no problem

closed 2022-08-06 by classicrocker883

classicrocker883, 2022-08-07:
This is the change I suggest for anyone who would like to choose either or
### _in Config_adv.h   --> Line 1375_

  //Change Title Menu Text to Centered
  #define TITLE_CENTERED  

### _in menus.cpp   --> Line 50 under void Draw_Title(TitleClass* title)_

 #if ENABLED(TITLE_CENTERED) 
  DWINUI::Draw_CenteredString(false, DWIN_FONT_HEAD, HMI_data.TitleTxt_color, HMI_data.TitleBg_color, (TITLE_HEIGHT - DWINUI::fontHeight(DWIN_FONT_HEAD)) / 2 - 1, title->caption);

  #else
  DWIN_Draw_String(false, DWIN_FONT_HEAD, HMI_data.TitleTxt_color, HMI_data.TitleBg_color, 14, (TITLE_HEIGHT - DWINUI::fontHeight(DWIN_FONT_HEAD)) / 2 - 1, title->caption);
#endif

mriscoc, 2022-08-21:
> @mriscoc This is the change I suggest for anyone who would like to choose either or
> 
> ### _in Config_adv.h --> Line 1375_
> //Change Title Menu Text to Centered #define TITLE_CENTERED
> 
I will to implement your contribution in a future version.