mriscoc/Ender3V2S1 #271: ABL routine fails on right edge after mesh got tilted

BogusF, 2022-08-04:
I've got a 9x9 mesh, HS probing disabled, multi probing disabled (one probe per meshpoint). I'm running [this ](https://github.com/mriscoc/Special_Configurations/releases/tag/AM3950T13) precompiled version on an E3 S1 pro, F1 Board. My probe has been relocated to be in line with nozzle on y-axis. On a sidenote, the physical machine limits are/were incorrect (too small) and I had to set correct values, mesh inset is close to maximum size from what the probe can reach, which is bigger than on stock since I relocated the probe. (x min=10,0; x max=216,0; y min 10,0; y max 218,0)

The unedited mesh is too far away on the right back corner so I tried the "tilt mesh" function which resulted in this weird jump over a nonexistant bump. I included a video because it is so ridiculous. It didn't do it on the untilted mesh. The gcode file certainly has no Z-movements there. I've had this issue before, but the jump wasn't as pronounced as the one I got filmed here. I think back then, I didn't tilt the mesh, but used the "edit mesh" function, but I wasn't out to document a bug so steps might have been different. I've only ever seen this on the right edge, but maybe I didn't get close enough to the other edges. I had to rename the gcode file to csv, so it would upload. 

[Here's the video](https://www.youtube.com/watch?v=6tVOHhxyDxw)

this is my current mesh (after the tilt)
[image]

This is the testfile it happened with. (rename to *.gcode)
[CE3_SquareTest.gcode.csv](https://github.com/mriscoc/Ender3V2S1/files/9258559/CE3_SquareTest.gcode.csv)

mriscoc, 2022-08-04:
Please, can you dump your mesh values?
Maybe with `G29 S-1`, `G29 T` or `M420 V`

BogusF, 2022-08-05:
I don't think the bug is specific to my mesh data, but here it is:

```
SENDING:G29 S-1
echo:  G29 I999
echo:  M421 I0 J0 Z0.0420
echo:  M421 I0 J1 Z0.1300
echo:  M421 I0 J2 Z0.0610
echo:  M421 I0 J3 Z0.0580
echo:  M421 I0 J4 Z0.0620
echo:  M421 I0 J5 Z0.0270
echo:  M421 I0 J6 Z0.0410
echo:  M421 I0 J7 Z0.0250
echo:  M421 I0 J8 Z0.0050
echo:  M421 I1 J0 Z0.0410
echo:  M421 I1 J1 Z0.1370
echo:  M421 I1 J2 Z0.0560
echo:  M421 I1 J3 Z0.0700
echo:  M421 I1 J4 Z0.0470
echo:  M421 I1 J5 Z0.0080
echo:  M421 I1 J6 Z0.0240
echo:  M421 I1 J7 Z0.0030
echo:  M421 I1 J8 Z-0.0170
echo:  M421 I2 J0 Z-0.0060
echo:  M421 I2 J1 Z0.0740
echo:  M421 I2 J2 Z0.0500
echo:  M421 I2 J3 Z0.0880
echo:  M421 I2 J4 Z0.0770
echo:  M421 I2 J5 Z0.0430
echo:  M421 I2 J6 Z-0.0010
echo:  M421 I2 J7 Z-0.0010
echo:  M421 I2 J8 Z-0.0390
echo:  M421 I3 J0 Z0.0380
echo:  M421 I3 J1 Z0.0670
echo:  M421 I3 J2 Z0.0610
echo:  M421 I3 J3 Z0.0770
echo:  M421 I3 J4 Z0.0470
echo:  M421 I3 J5 Z0.0410
echo:  M421 I3 J6 Z0.0210
echo:  M421 I3 J7 Z0.0010
echo:  M421 I3 J8 Z-0.0070
echo:  M421 I4 J0 Z0.0180
echo:  M421 I4 J1 Z0.0810
echo:  M421 I4 J2 Z0.0420
echo:  M421 I4 J3 Z0.0920
echo:  M421 I4 J4 Z0.0430
echo:  M421 I4 J5 Z0.0510
echo:  M421 I4 J6 Z0.0110
echo:  M421 I4 J7 Z0.0100
echo:  M421 I4 J8 Z0.0240
echo:  M421 I5 J0 Z0.0330
echo:  M421 I5 J1 Z0.0670
echo:  M421 I5 J2 Z0.0410
echo:  M421 I5 J3 Z0.0670
echo:  M421 I5 J4 Z0.0590
echo:  M421 I5 J5 Z0.0410
echo:  M421 I5 J6 Z0.0180
echo:  M421 I5 J7 Z0.0050
echo:  M421 I5 J8 Z-0.0050
echo:  M421 I6 J0 Z0.0890
echo:  M421 I6 J1 Z0.1340
echo:  M421 I6 J2 Z0.0950
echo:  M421 I6 J3 Z0.0970
echo:  M421 I6 J4 Z0.0730
echo:  M421 I6 J5 Z0.0600
echo:  M421 I6 J6 Z0.0470
echo:  M421 I6 J7 Z0.0320
echo:  M421 I6 J8 Z0.0330
echo:  M421 I7 J0 Z0.0750
echo:  M421 I7 J1 Z0.1010
echo:  M421 I7 J2 Z0.0430
echo:  M421 I7 J3 Z0.0660
echo:  M421 I7 J4 Z0.0210
echo:  M421 I7 J5 Z0.0120
echo:  M421 I7 J6 Z0.0130
echo:  M421 I7 J7 Z0.0020
echo:  M421 I7 J8 Z0.0040
echo:  M421 I8 J0 Z0.0800
echo:  M421 I8 J1 Z0.1230
echo:  M421 I8 J2 Z0.0700
echo:  M421 I8 J3 Z0.1040
echo:  M421 I8 J4 Z0.0570
echo:  M421 I8 J5 Z0.0440
echo:  M421 I8 J6 Z0.0470
echo:  M421 I8 J7 Z0.0290
echo:  M421 I8 J8 Z0.0260
```

One other thing I found out is, the bug disappears after a reboot. I took the mesh posted above, had it tilted, tried a print, bug was there. Stored the mesh to memory slot#1, rebooted the printer and loaded mesh from said slot, tried a print, bug was gone.

One possible source might be that on the right side of the bed, the mesh doesn't cover the whole bed since the probe is offset to the left. So the spot where the bug happens is outside the mesh area. For people with stock probe position with offset in x _and_ y, the bug might happen in the back also since they can't probe that far in the back. I'll try a smaller mesh inset, see if the bug appears at other spots as well.

BogusF, 2022-08-05:
Okay, reducing the mesh inset yielded results:

on a 5x5 mesh, it said "the print got stopped" and I had to confirm to continue, but there was no "jump" bug.
On a 9x9 mesh I had the same print got stopped confirmation thing, after confirming it continued printing, left and back side no jumps, on the right side, the jump was so high, the printhead went outside of printing area and I had to shut down the printer to prevent it from damaging itself. (skipping z-stepper motors)
Didn't test any further since I don't want to damage my printer.

So it does have to do with printing outside the mesh area, but it still only happens on the right side. It can be avoided with printing inside mesh area and, if tilt function has to be used, you should reboot the printer afterwards to be safe.

mriscoc, 2022-08-06:
Please, I want to have the mesh data already tilted to see if there is any anomalous value.

mriscoc, 2022-08-06:
>the printhead went outside of printing area 

You can set the physical limits to avoid that.

BogusF, 2022-08-06:
As I wrote, I didn't get the jump on a 5x5 mesh. Also, my physical limits are correctly set. Manually, it's impossible to go out of printing area. At least that's how it was, when I set it up. Will check again.

I'll have my mesh tilted and get the data dumped, but as I wrote, if i store the tilted mesh to a memory slot and reboot, load said mesh after the reboot, the problem disappears.

BogusF, 2022-08-06:
I confirmed, my machine physical limits are correctly set. Here's the dump after tilting command:

```
SENDING:G29 S-1
echo:  G29 I999
echo:  M421 I0 J0 Z0.0426
echo:  M421 I0 J1 Z0.1355
echo:  M421 I0 J2 Z0.0714
echo:  M421 I0 J3 Z0.0733
echo:  M421 I0 J4 Z0.0822
echo:  M421 I0 J5 Z0.0522
echo:  M421 I0 J6 Z0.0711
echo:  M421 I0 J7 Z0.0600
echo:  M421 I0 J8 Z0.0449
echo:  M421 I1 J0 Z0.0381
echo:  M421 I1 J1 Z0.1390
echo:  M421 I1 J2 Z0.0629
echo:  M421 I1 J3 Z0.0818
echo:  M421 I1 J4 Z0.0637
echo:  M421 I1 J5 Z0.0296
echo:  M421 I1 J6 Z0.0505
echo:  M421 I1 J7 Z0.0344
echo:  M421 I1 J8 Z0.0193
echo:  M421 I2 J0 Z-0.0124
echo:  M421 I2 J1 Z0.0725
echo:  M421 I2 J2 Z0.0534
echo:  M421 I2 J3 Z0.0963
echo:  M421 I2 J4 Z0.0902
echo:  M421 I2 J5 Z0.0611
echo:  M421 I2 J6 Z0.0220
echo:  M421 I2 J7 Z0.0269
echo:  M421 I2 J8 Z-0.0062
echo:  M421 I3 J0 Z0.0280
echo:  M421 I3 J1 Z0.0619
echo:  M421 I3 J2 Z0.0608
echo:  M421 I3 J3 Z0.0817
echo:  M421 I3 J4 Z0.0567
echo:  M421 I3 J5 Z0.0556
echo:  M421 I3 J6 Z0.0405
echo:  M421 I3 J7 Z0.0254
echo:  M421 I3 J8 Z0.0223
echo:  M421 I4 J0 Z0.0045
echo:  M421 I4 J1 Z0.0724
echo:  M421 I4 J2 Z0.0383
echo:  M421 I4 J3 Z0.0932
echo:  M421 I4 J4 Z0.0491
echo:  M421 I4 J5 Z0.0620
echo:  M421 I4 J6 Z0.0269
echo:  M421 I4 J7 Z0.0308
echo:  M421 I4 J8 Z0.0497
echo:  M421 I5 J0 Z0.0160
echo:  M421 I5 J1 Z0.0549
echo:  M421 I5 J2 Z0.0338
echo:  M421 I5 J3 Z0.0647
echo:  M421 I5 J4 Z0.0616
echo:  M421 I5 J5 Z0.0485
echo:  M421 I5 J6 Z0.0304
echo:  M421 I5 J7 Z0.0223
echo:  M421 I5 J8 Z0.0172
echo:  M421 I6 J0 Z0.0684
echo:  M421 I6 J1 Z0.1183
echo:  M421 I6 J2 Z0.0842
echo:  M421 I6 J3 Z0.0912
echo:  M421 I6 J4 Z0.0721
echo:  M421 I6 J5 Z0.0640
echo:  M421 I6 J6 Z0.0559
echo:  M421 I6 J7 Z0.0458
echo:  M421 I6 J8 Z0.0517
echo:  M421 I7 J0 Z0.0509
echo:  M421 I7 J1 Z0.0818
echo:  M421 I7 J2 Z0.0287
echo:  M421 I7 J3 Z0.0566
echo:  M421 I7 J4 Z0.0165
echo:  M421 I7 J5 Z0.0124
echo:  M421 I7 J6 Z0.0183
echo:  M421 I7 J7 Z0.0122
echo:  M421 I7 J8 Z0.0192
echo:  M421 I8 J0 Z0.0524
echo:  M421 I8 J1 Z0.1003
echo:  M421 I8 J2 Z0.0522
echo:  M421 I8 J3 Z0.0911
echo:  M421 I8 J4 Z0.0490
echo:  M421 I8 J5 Z0.0409
echo:  M421 I8 J6 Z0.0488
echo:  M421 I8 J7 Z0.0357
echo:  M421 I8 J8 Z0.0376
```

Since the user prompt is something the printer doesn't do on normal prints, here's the the console output of the printer after starting the print:

```
echo:SD card ok
echo:Now fresh file: CE3_SQ~1.GCO
File opened: CE3_SQ~1.GCO Size: 69381
File selected
echo:Now fresh file: ce3_sq~1.gco
File opened: ce3_sq~1.gco Size: 69381
File selected
//action:resume
//action:prompt_end
//action:prompt_begin Resuming SD
//action:prompt_button Dismiss
//action:prompt_show
//action:prompt_end
//action:prompt_begin M1 Stop
//action:prompt_button Continue
//action:prompt_show
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:Bed Leveling ON
echo:Fade Height 2.00
//action:cancel
```
It did jump again, but since I went back to the original bigger mesh inset, it stayed inside the printable area. After a reboot, it didn't prompt for user interaction after starting the print and it didn't jump anymore. 

If you want me to perform any other tests, it should be today, I'm not gonna have access to my printer for 4 weeks starting tomorrow.

mriscoc, 2022-08-06:
Please, post a capture of your info page. Thanks

VivantSenior, 2022-08-13:
I have the same bug. "Jump" happens in back right corner of the bed. My BLTouch is on the left hand side from the nozzle. HS mode enabled.

Probe offsets:
X: -46.0
Y: -19.0
Z: -2.53

Bed size:
X: 230
Y: 230

Min position:
X: 0
Y: 0

Max position:
X: 248
Y: 231
Z: 250

mriscoc, 2022-08-14:
Thank you for your report, I'm working on trying to replicate the bug to fix it.

thinkyhead, 2022-08-14:
UBL has some anomalies with move segmentation that can be solved by enabling SEGMENT_LEVELED_MOVES. And, when "outside" the known mesh bounds there may be a jump in Z with UBL. Perhaps this is a manifestation of one of those issues.

mriscoc, 2022-08-14:
> UBL has some anomalies with move segmentation that can be solved by enabling SEGMENT_LEVELED_MOVES. And, when "outside" the known mesh bounds there may be a jump in Z with UBL. Perhaps this is a manifestation of one of those issues.

Thanks. I was able to replicate the bug behavior, I will try those fixes.

mriscoc, 2022-08-14:
How to replicate this bug:
```gcode
M502          ; Reset to defaults
C29 N9 T0   ; Mesh size to 9x9 and Pre-heat to 0
C851 M0     ; Multiple probing to 0
G29 P1        ; Take a new mesh
G29 A         ; Active the mesh leveling
G0 X205 Y140 ; move to the mesh border
G0 Z0         ; Move Z to 0, the head moves to the expected position
G0 X210 Y140 ; exceed the mesh border, the head moves and Z is raised
G0 X205 Y140; return to the border, The head moves to the expected position
```
This behavior was not possible to replicate with a mesh lower than 9x9.

mriscoc, 2022-08-14:
With SEGMENT_LEVELED_MOVES disabled, the bug doesn't occur.

mriscoc, 2022-08-14:
A PR was submitted to the Marlin repository fixing this issue: https://github.com/MarlinFirmware/Marlin/pull/24631 and will be available in the next firmware release.

limamg, 2022-08-16:
> > A UBL tem algumas anomalias com segmentação de movimento que podem ser resolvidas ativando SEGMENT_LEVELED_MOVES. E, quando "fora" dos limites de malha conhecidos, pode haver um salto em Z com UBL. Talvez esta seja uma manifestação de uma dessas questões.
> 
> Obrigado. Consegui replicar o bug, o mesmo comportamento foi encontrado no atual bugfix Marlin 2.1.x. Vou tentar essas correções.
> 
> EDIT: SEGMENT_LEVELED_MOVES estava ativo em Configuration.h

do you think it would be interesting to modify the file and compile the firmware at this point?
The mesh I use is 5x5

> > UBL has some anomalies with move segmentation that can be solved by enabling SEGMENT_LEVELED_MOVES. And, when "outside" the known mesh bounds there may be a jump in Z with UBL. Perhaps this is a manifestation of one of those issues.
> 
> Thanks. I was able to replicate the bug, the same behavior was found in the current Marlin bugfix 2.1.x. I will try those fixes.
> 
> EDIT: SEGMENT_LEVELED_MOVES was active in Configuration.h

do you think it would be interesting to modify the file and compile the firmware at this point?
The mesh I use is 5x5

mriscoc, 2022-08-16:
The next version will have this issue fixed.

mriscoc, 2022-08-17:
Fixed in the latest version.

closed 2022-08-17 by mriscoc