mriscoc/Ender3V2S1 #277: Have Sound(beep) disabled for button clicks, but leave enabled when printing (with Gcode)

classicrocker883, 2022-08-06:
When I disable sound, it disables it entirely.

I'd like it to just be disabled for button clicks - Menu navigation. 

Is there a way to leave it enabled when printing, say for Gcode to produce?

mriscoc, 2022-08-06:
It is possible to mute menu clicks by editing the file enconder.cpp and comment out the content of `Encoder_tick`:
https://github.com/MarlinFirmware/Marlin/blob/9c86ca3a19e3c557feabbd0e4167372e950553d8/Marlin/src/lcd/e3v2/common/encoder.cpp#L52-L54

classicrocker883, 2022-08-07:
@mriscoc 
I commented out 
>**Encoder_Tick();** 

on line 88 in encoder.cpp
i didnt actually try commenting out 

> ```c++
> void Encoder_tick() {
>   #if PIN_EXISTS(BEEPER)
>     if (ui.sound_on) buzzer.click(10);
>   #endif
> }
> ```

### i also found a bug

file:  src/inc/Conditionals_Adv.h
line:  1028

it should be

 > **#ifndef LCD_SERIAL_PORT**

originally it is **#ifdef**
because the LCD serial port for ender3v2 and the voxelab aquila is defined as >> **3**

the warning shows up when line 105 is commented out in Configuration.h

>//#define NO_AUTO_ASSIGN_WARNING

mriscoc, 2022-08-08:
The line:
```cpp
  #ifdef LCD_SERIAL_PORT
```
Is correct, see: https://github.com/MarlinFirmware/Marlin/blob/5b68a3f79b845c4c125c43cfd0d506073da6007e/Marlin/src/inc/Conditionals_adv.h#L1049

The line:
```cpp
#define NO_AUTO_ASSIGN_WARNING
``` 
Is there to avoid the warnings because LCD_SERIAL_PORT was defined automatically in Conditionals_Adv.h

mriscoc, 2022-08-08:
This is not a firmware issue.

closed 2022-08-08 by mriscoc