mriscoc/Ender3V2S1 #280: V2 Neo support

adrianocr, 2022-08-09:
I'm wondering if the new V2 refresh ("neo") can be added to this project. It comes with auto leveling and some slightly upgraded internal bits too and a new screen but I think the mainboard is the same as the standard V2.

mriscoc, 2022-08-09:
Test with the Ender3V2-422-BLTUBL firmware.

adrianocr, 2022-08-09:
@mriscoc that actually worked great! The only concern so far is that I'm not sure I got the physical settings correct. Do you have any tips on how to find the correct values? Once again it seems that the V2 neo is so recent that a google search came up blank.

The firmware seems to start off at 230x230x250 but the V2 Neo is actually 220x220x250mm so I adjusted that and subsequently tried manually moving the axes around and came up with x min -26, y min -26, x max 218, y max 205, and z max 250. Additionally I had to manually set the prove offsets to x -21 and y (+)7.

For anyone in the future that sees this: those values may be slightly incorrect.

mriscoc, 2022-08-09:
Thanks for your report, XY MIN/MAX Position and other physical settings can currently be found only experimenting like you did.

ssrshark, 2022-08-10:
Ender 3 V2 Neo now has made the thumbnail preview work on DACAI screen, It is very cool if you add this to your project. This is what of us have been waiting for.

mriscoc, 2022-08-10:
Are you sure that they are DACAI? I did receive reports that they are using a fully compatible with DWIN screens.

adrianocr, 2022-08-10:
@mriscoc it does seem that my unit also has a DACAI screen but this firmware with the DWIN files (to the best of my understanding) works flawlessly too.

ssrshark, 2022-08-10:
I owed Ender V2 Neo, the screen they used is DACAI, I update screen firmware with Private not DWIN.

mriscoc, 2022-08-10:
> @mriscoc it does seem that my unit also has a DACAI screen but this firmware with the DWIN files (to the best of my understanding) works flawlessly too.

Is the Professional firmware showing the preview when you try to print this demo G-codes? :
[demo.zip](https://github.com/mriscoc/Ender3V2S1/files/9295457/demo.zip)

mriscoc, 2022-08-10:
> 

> I owed Ender V2 Neo, the screen they used is DACAI, I update screen firmware with Private not DWIN.

Good to know, so they update the firmware for the DACAI screens, please test with the provided demo G-codes and Professional Firmware.

ssrshark, 2022-08-10:
Professional firmware for 4.22 board, shown blank while preview on screen, we have to use their 4.2.2 firmware and the "Private" for screen which Creality privide to in order to make it work, I think if you check for it you would make it work.

https://www.creality.com/pages/download-ender-3-v2-neo?spm=..page_1934481.products_display_1.1&spm_prev=..page_1934087.header_1.1

mriscoc, 2022-08-10:
Only binary files were published, we need to have the official sources or the API information for accessing the newly implemented functions on the DACAI new firmware.

The users of that printer can request to Creality to disclosure the Official source code for the Neo printer.

adrianocr, 2022-08-10:
>The users of that printer can request to Creality to disclosure the Official source code for the Neo printer.

By email or is there a special request process or something? I wouldn't mind doing it.

mriscoc, 2022-08-10:
> By email or is there a special request process or something? I wouldn't mind doing it.

Use all the channels as possible, mail to support, request in the [Creality repository](https://github.com/CrealityOfficial), etc.

ssrshark, 2022-08-10:
I have contacted them via email couple day ago about the instruction how to generate thumbnail preview on screen, but they provided me with the useless instruction, but I have found that to generate preview thumbnail they use this slicer "
Creality_Print" "https://www.crealitycloud.com/software-firmware/software" there is an option in printer manager to generate thumbnail to Gcode.

adrianocr, 2022-08-11:
@mriscoc so I'm having an issue where it seems I am incapable of getting the bed properly centered. Any tips?

I've set the bed size to 220x220 (even though if you actually measure it with something it is actually around 235x235) and I have manually tested all the end stops to come up with the X max and Y max, and manually centered the probe offsets by placing a dot in the direct center of the bed with a sharpie and then painstakingly centering the probe until it perfectly touches that center point (interestingly here the measurements I get with my caliper are off by 3mm on the x axis and a whole 11mm on the y axis). Auto home seems to work correctly and the device perfectly probes the direct center of the bed during homing.

However when I go print something the prints are always about 10-15cm left of center. I've tried prusa slicer and cura and manual gcodes. 

Any suggestions on what could be going on and how I could sort it out?

mriscoc, 2022-08-11:
If auto home goes to the center, it is very possible that the problems are in the slicer settings. If you can send manual G-codes, if your bed is 220x220 send `G0 X110 Y110` the nozzle will go to the center of the bed if the printer settings are correct.

MarkforU, 2022-08-12:
You can make a center line on your bed, there's a file on Thingiverse to print a target. You adjust CURA to be center by going into Machine Settings and adjusting extruder in there for the amount more or less that you need to be on either the X and or Y Axis. That's where your adjustments take place for center printing. There's a YouTube video on this exact problem. I ran into it with Jyers and fixes it then, haven't paid attention to the new software and I'm using a printable area of 230x230mm which I set both in Cura and also the printer. I also limit my z-height also.

DynaMight1124, 2022-08-20:
I came here because I have a S1 with a DACAI screen, I saw the V2 Neo had thumbnail support. I went to the Creality website and downloaded the V2 Neo firmware, I noticed it had the Private folder indicating a DACAI screen. I flashed the screen firmware and it flashed fine, the Professional firmware works fine. It had an actual firmware update called "firmware.zlib" which I hadnt seen before. I assume this is where the magic is. 

The thumbnails didnt work however but it does seem the screen firmware has been updated to allow it. It just needs people cleverer than me to work out how :)

curather, 2022-08-25:
Tagging the link to Creality V2 Neo issue to get [Support the V2 Neo #1](https://github.com/CrealityOfficial/Ender-3V2/issues/1)

mriscoc, 2022-08-31:
Please, can someone post here a file that the Neo uses to show the preview on the screen?

adrianocr, 2022-09-01:
@mriscoc I'm not sure I understood correctly but here are the two files that come on the V2 Neo's SD card: 
[Archive.zip](https://github.com/mriscoc/Ender3V2S1/files/9472671/Archive.zip)

Both of those show a preview on the DACAI screen if you run the print via the Neo's screen.

mriscoc, 2022-09-01:
Interesting, thank you!

orifer, 2022-09-02:
[image]
[image]

So I installed the Neo firmware + screen and tried to load these files in my V2 with a DACAI screen and the thumbnails loaded. However, it's not loading when trying with a generated one with the python script. I saw some slight changes on the jpeg base64 encoding, but couldn't figure it out.

mriscoc, 2022-09-03:
I can now also open the preview of the Neo files but only with DWIN displays:
  
[image][image]
  
We need the DACAI API to enable preview on that screen.

curather, 2022-09-19:
firmware.zlib does upgrade the firmware on the DACAI display.  I have a ender 3 v2.  I tested with the two different gcodes(one from manufactorer and other from cura), neither worked so it definitely has to do with the api calls to the display.