mriscoc/Ender3V2S1 #285: Bed leveling wtith 9x9 mesh fails

od1001, 2022-08-12:
Ender3S1-F1-20220716.bin hat einen Bug. Bed-Leveling mit 9x9 Mesh funktioniert nicht. Vor Probe Point 67 wird der Taststift eingefahren, sodass die Abtastung fehlschlägt. Das Bed-Leveling wird dann wiederholt. Sieht nach Speicherüberschreiber aus.

Ender3S1-F1-20220716.bin has a bug. Bed leveling with 9x9 mesh does not work. Before the Point 67 trial, the tape pencil is run in so that the scanning fails. The Bed leveling is then repeated. Looks like storage overwriters.

mriscoc, 2022-08-12:
DIsable HS mode: https://github.com/mriscoc/Ender3V2S1/wiki/3D-BLTouch

od1001, 2022-08-12:
I have an ender-3 s1 with no BLTouch.

mriscoc, 2022-08-12:
Please search about CRTouch and BLTouch comparisons

closed 2022-08-12 by mriscoc