mriscoc/Ender3V2S1 #288: S1 Pro filament sensor does not work

acevalsl, 2022-08-13:
**Describe the bug**
Cuando instalo el Firmware nuevo el sensor de filamento deja de funcionar.

Tenga o no tenga filamento la impresora sigue funcionando. Cuando cargo otra vez el Firmware original funciona perfectamente.

Tener en cuenta que la S1 pro lleva un sensor de filamento óptico.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version (please complete the following information):**
Ender3S1-F4-T13-20220717
Ender3S1-F4-20220716

**Additional context**
Add any other context about the problem here.

acevalsl, 2022-08-13:
https://user-images.githubusercontent.com/22813078/184509500-881ae516-ab06-4999-b00b-6f78937ccd8f.mov

mriscoc, 2022-08-13:
Did you check your runout distance?
Use the End-stop diagnostic to set up your runout sensor. The image is when there is filament inside of the sensor:
[image]

acevalsl, 2022-08-13:
Hola, en esa pantalla veo que tanto cuando tengo el filamento como cuando lo quito responde perfectamente.

Sin embargo luego cuando esté imprimiendo lo que te lo ponga siempre imprime.

Lo que me ha llamado la atención es que el z_min aparece Triggered y no sé por qué?
He intentado pasarte un vídeo pero dice que es demasiado grande.

Hello, on that screen I see that both when I have the filament and when I remove it it responds perfectly.

However, when you are printing what you put it on, always print.

What has caught my attention is that the z_min appears Triggered and I don't know why?

I've tried to send you a video but it says it's too big.

acevalsl, 2022-08-13:
https://user-images.githubusercontent.com/22813078/184510413-fe79c48c-c941-4af9-8c30-99158f052565.mov

mriscoc, 2022-08-13:
> IMG_4940-1.mov

The firmware recognizes your sensor correctly, please read about runout distance, or look for G-codes disabling the runout sensor.

acevalsl, 2022-08-13:
Activated the code M 412 S1 to test it by typing it after the code M420 in the gcode.

If you have any idea, because I don't know how to activate it, I thought it was with that code.

acevalsl, 2022-08-13:
I just tried M412 D10 and the sensor has worked. Thanks a lot

closed 2022-08-13 by mriscoc

acevalsl, 2022-10-11:
https://drive.google.com/drive/folders/1-cpvnjTAg99hUYbaq6m5Ycl3gjZOLGy4?usp=sharing <https://drive.google.com/drive/folders/1-cpvnjTAg99hUYbaq6m5Ycl3gjZOLGy4?usp=sharing>

Video