mriscoc/Ender3V2S1 #290: Auto Build Mesh on  BTT SKR Mini E3 V3 board No Response

haqbany, 2022-08-17:
Auto Build Mesh on  BTT SKR Mini E3 V3 board No Response 

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version (please complete the following information):**
Firmware version and release date

**Additional context**
Add any other context about the problem here.

haqbany, 2022-08-17:
as will nothing respond after the update , and the printer make little strange fan sound !?

mriscoc, 2022-08-18:
Is the previous version https://github.com/mriscoc/Ender3V2S1/releases/download/20220527/Ender3V2-SKRMINI-BLTUBL-20220527.bin working for you?

haqbany, 2022-08-18:
Yes was working fine but after the update this happened

mriscoc, 2022-08-18:
Could you test again please for discard problems with hardware?

haqbany, 2022-08-18:
Hi Miguel , 

when i use > Tilt Mesh , it working ok 
Auto Build Mesh no response 
Bed traming ok 

here is a video link 
https://youtu.be/01dDVfYa9BE
Best Regards,
Rashid

haqbany, 2022-08-18:
also i would like to contribute with translation in to arabic i know little about programing but i'm sure i can do the translation if you're interested. as will graphic design .

mriscoc, 2022-08-18:
From your video, something is not ok with your thermistors and/or heaters. The error is in the bed pre-heating.
In Advanced Settings / Mesh leveling / Set bed temperature to 0 and try again.

haqbany, 2022-08-18:
thermistors is not connected, i have Set bed temperature to 0 and try again same thing 
.

mriscoc, 2022-08-18:
> also i would like to contribute with translation in to arabic i know little about programing but i'm sure i can do the translation if you're interested. as will graphic design .

The DWIN/DACAI screens have a very limited charset mainly English and Chinese.

mriscoc, 2022-08-18:
> thermistors is not connected, i have Set bed temperature to 0 and try again same thing .

It is difficult to predict what will be the behavior when the thermistor is disconnected, but I expected to have a runaway thermal alert.

haqbany, 2022-08-21:
Thank you , for the update you are right i had replace thermistor and thing working great

haqbany, 2022-08-21:
Can you make the firmware compatible  with touch screen from big treetouch ? many people has has purchased this screen but the firmware from them is horrible , if i need to use your firmware now witch in my opinion the best so far I will have to go back to the old screen. If you can do that I we all appreciate it

mriscoc, 2022-08-22:
Sorry, but touchscreens are totally different from the screens that I have.  I only code for hardware similar to the one I have.

closed 2022-08-22 by mriscoc

Nbrooks639, 2022-08-22:
> Can you make the firmware compatible with touch screen from big treetouch ? many people has has purchased this screen but the firmware from them is horrible , if i need to use your firmware now witch in my opinion the best so far I will have to go back to the old screen. If you can do that I we all appreciate it the screen link https://biqu.equipment/collections/lcd/products/btt-tft35-e3-v3-0-display-touch-screen-two-working-modes

you can hook up a ender v2/s1 screen to the skr board, it also just got even easier because BIQU made a cable

https://biqu.equipment/products/mini-e3-connecting-cable-for-ender-3-v2-display-screen?_pos=4&_sid=ff81c529c&_ss=r

[image]

haqbany, 2022-10-11:
Hi ,yes it was working just fine
Regards