mriscoc/Ender3V2S1 #293: Firmware update take too much time

hkpmatt, 2022-08-19:
I have updateto mm and bought a 3d touch and try to update to blt version.
my board is 4.2.2 and when the sd card is insert and the printer is turn on.
it pid keep stay blank and wont update the firmware.
any ideas why it happen?

limamg, 2022-08-19:
> I have updateto mm and bought a 3d touch and try to update to blt version. my board is 4.2.2 and when the sd card is insert and the printer is turn on. it pid keep stay blank and wont update the firmware. any ideas why it happen?

test it by changing the file name because it doesn't have to be a different name from the ones you've used before

mriscoc, 2022-08-19:
https://github.com/mriscoc/Ender3V2S1/wiki/How-to-install-the-firmware

mriscoc, 2022-08-22:
No more user feedback.

closed 2022-08-22 by mriscoc

hkpmatt, 2022-08-26:
I tried to format the SD and sometimes it still get blank