mriscoc/Ender3V2S1 #296: When octoprint is connected, it gives error code M112.

onur94, 2022-08-19:
**Describe the bug**
When printing starts while octoprint is connected, it gives error code M112.

**Screenshots**
https://ibb.co/2MLz5L6
https://ibb.co/HCRNhXM

I connected it to the octoprint to monitor temperature and percentage values. After the bed and nozzle reached the required temperature, I started printing. As soon as the nozzle touches the bed, it gives M112 printer killed error. It didn't do this in the previous version.

**Version (please complete the following information):**
20220816

DokuKaefer, 2022-08-19:
See here:

https://github.com/mriscoc/Ender3V2S1/wiki/Octoprint

mriscoc, 2022-08-19:
Please read the Wiki: https://github.com/mriscoc/Ender3V2S1/wiki/Octoprint#error-handling

closed 2022-08-19 by mriscoc

onur94, 2022-08-19:
> Please read the Wiki: https://github.com/mriscoc/Ender3V2S1/wiki/Octoprint#error-handling

I fixed it from where you pointed, thanks.