mriscoc/Ender3V2S1 #301: Sprite Extruder Pro Kit + CR-Touch: leveling outside of the bed

CafeRacer400, 2022-08-22:
**Describe the bug**
I've got a **Ender3V2** and mounted the new **Creality Sprite Extruder Pro Kit** onto it. In addition to that, im using a **CR-Touch** bed level sensor which can be mounted on the Sprite Extruder since there is a braket extra for it.
When i want to create a **bed mesh** the Extruder-Head **moves too far in the front** that the **CR-Touch sensor is arround 1-1,5cm outside the bed** and wanna do a level there _(this will cause a crash of the nozzle with the heatbed ofc!)_

**To Reproduce**
Steps to reproduce the behavior:
1. Use the Creality Sprite Extruder Pro Kit with an CR-Touch
2. Click on Prepare
3. Auto Build Mesh
4. See a coming crash :)

**Expected behavior**
Can you change the coordinates of the auto build mesh special for CR/BL-Touch used with the Sprite Extruder Pro Kit?

**Screenshots**
[image]

**Version:**
Firmware Ender3V2-422-BLT-20220816.bin

mshihinski, 2022-08-22:
@CafeRacer400 try using  -36.5x and -40y for the probe offsets under advanced settings.

mriscoc, 2022-08-22:
Check: https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides

mriscoc, 2022-08-27:
No more feedback

closed 2022-08-27 by mriscoc