mriscoc/Ender3V2S1 #304: Example BLTouch Marlin start G Code loops probing sequence or crashes Ender 3v2 4.2.2

Router-Jockey, 2022-08-23:
### Discussed in https://github.com/mriscoc/Ender3V2S1/discussions/303

<div type='discussions-op-text'>

<sup>Originally posted by **Router-Jockey** August 22, 2022</sup>
I tried the August maintenance build for Ender 3 v2 4.2.2 BL Touch and Unified. 
BLTouch version caused loops when sending;

G28           ; Home XYZ.
G29 P1        ; Do automated probing of the bed.
G29 P3        ; Smart Fill Repeat until all mesh points are filled in, Used to fill unreachable points.
G29 S0        ; Save UBL mesh points to slot 0 (EEPROM).
G29 F 10.0    ; Set Fade Height for correction at 10.0 mm.
G29 A         ; Activate the UBL System.

The August maintenance Unified Bed Leveling build worked better but X-Axis stopped working when printing. Probing went fine but when it went to run the priming strip down the side of the bed surface it stayed in the same X position and made a mess.

I restored defaults between each flash, no change. Went back to the previous maintenance release Ender3V2-422-BLTUBL-20220527.bin and everything works again. At least I know my printer isn't the reason. :) 

Some other things worth noting although trivial were the click-wheel sensitivity is VERY low on the August maintenance build which makes it take forever to reach 433 e-steps from the default 93 - need to set this for my Sprite Pro extruder.

Happy to do some more testing if it helps. </div>

mriscoc, 2022-08-23:
I did your proposed command with the August Ender3V2-422-BLTUBL version, all was working fine. If your X-axis stopped that can be caused by overheating due to the higher speeds of the firmware.

```
Send: C29 T0 N3
Recv: echo:; Mesh Insets and leveling settings:
Recv: echo:  C29 L25.00 R205.00 F25.00 B205.00 N3 T0 ; 3x3  T=0°C
Recv: ok
Send: G28
Recv: ok
Send: G29 P1
Recv: Default storage slot 0 selected.
Recv: Mesh invalidated. Probing mesh.
Recv: Probing mesh point 1/9.
Recv: Probing mesh point 2/9.
Recv: Probing mesh point 3/9.
Recv: Probing mesh point 4/9.
Recv: Probing mesh point 5/9.
Recv: Probing mesh point 6/9.
Recv: Probing mesh point 7/9.
Recv: Probing mesh point 8/9.
Recv: Probing mesh point 9/9.
Recv: ok
Send: G29 P3
Recv: ok
Send: G29 S0
Recv: Mesh saved in slot 0
Recv: Done.
Recv: ok
Send: G29 F 10.0
Recv: ok
Send: G29 A
Recv: Unified Bed Leveling System v1.01 active
Recv: ok
```

Router-Jockey, 2022-08-23:
OK I'll try omitting the G29 S1 and see if it works.
I'm not sure it was overheating... it had just finished building the mesh immediately before with the nozzle at 175 then paused at the start position to heat to 205, made the first priming line and didn't move over at all for the second.
Then instead of lifting the nozzle to move to the start of the print it just moved along the Y axis where I would expect it to print. Did that until I could stop it.

Like I said, I'll try the August release again with BTUBL and report back.

Thank you!

mriscoc, 2022-08-31:
No more user feedback

closed 2022-08-31 by mriscoc