mriscoc/Ender3V2S1 #305: LCD Landscape

mattkerrison, 2022-08-23:
### Discussed in https://github.com/mriscoc/Ender3V2S1/discussions/102

<div type='discussions-op-text'>

<sup>Originally posted by **Frohmut95** March 27, 2022</sup>
Hey, is there an Option to rotate the Standard Ender 3 V2 Display to landscape mode? in the Marlin Files where some snippets of code for this but couldnt find a way.</div>

Hey @mriscoc, awesome firmware - I had several issues with Jyers that seem to have been resolved by flashing your firmware.

I saw the linked discussion (#102) as well [as this link](https://otioss.com/posts/ender3-v2-front-panel), but still not 100% sure on how to make it work with this firmware.

Am I correct in thinking there is no need to flash my Ender 3 V2 display, but I need to compile a version of the Professional Firmware that has the flag `DWIN_MARLINUI_LANDSCAPE` enabled instead of `DWIN_LCD_PROUI`?

Will it reduce functionality using MarlinUI with Professional firmware? ie: does the DWIN_MARLINUI_LANDSCAPE still show all menu items etc?
Is there anything else that needs doing apart from changing that flag, like supporting files from Marlin that aren't in this repo already?

I really want to get landscape mode working as I use Octodash as well and want to integrate both into a front panel. Appriciate your help!

mriscoc, 2022-08-23:
HI,
The user interface of vanilla Marlin is MARLINUI
The old user interface of Creality is CREALITY
The user interface of Professional Firmware is PROUI
The user interface of Jyers firmware is JYERSUI

You can't mix UI.

A landscape version for the ProUI is not planned.

closed 2022-08-23 by mriscoc

classicrocker883, 2022-08-25:
**it is possible** to to have it turned landscape.
but it will take extensive dwin rework to reposition all the menu items.

### changing a single parameter will get you this:
[image]

limamg, 2022-08-25:
rotate the screen and not have it all filled and parts are missing, what's the use?

mriscoc, 2022-08-25:
> **it is possible** to to have it turned landscape. but it will take extensive dwin rework to reposition all the menu items.
> 
> ### changing a single parameter will get you this:
> [image]

Yes. I know, you can use DWIN_Frame_SetDir to change screen orientation, but it only changes how the display firmware draws the screen, actually it is not too much work, you must to change the DWINUI according to the new position and dimensions, and move the dash board.

mattkerrison, 2022-08-26:
Shame, the LCD panel itself looks like it's viewing angles are **much** better in landscape than portrait, was hoping it would be relatively achievable as the rest of the firmware is excellent