mriscoc/Ender3V2S1 #306: HAS_PROBE does not exist

classicrocker883, 2022-08-23:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### lcd\e3v2\proui\dwin,h
_line 230_

**originally**
#if !HAS_PROBE
  void HomeZandDisable();
#endif

### updated to

#if !HAS_BED_PROBE
  void HomeZandDisable();
#endif

//there is no such define HAS_PROBE
so it should be HAS_BED_PROBE

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version (please complete the following information):**
Firmware version and release date

**Additional context**
Add any other context about the problem here.

mriscoc, 2022-08-23:
Thanks for the bug catch.

classicrocker883, 2022-08-25:
@mriscoc 
im working on a few things, like adding the option for the BED_SCREW_INSET which changes the position where the nozzle or probe goes when leveling / tramming the corners.
also having the LCD_BACKLIGHT_TIMEOUT enabled with a menu option to change the value
is this a feature you look forward to having?

i am having trouble getting the lcd backlight timeout to work, i posted the issue under marlinfirmare.
so far it does work well just having the issue of resetting the timeout timer.

mriscoc, 2022-08-25:
You don't need to set the bed tramming points over the screws positions, it is better at the mesh corners, that minimizes the mesh compensation values.

classicrocker883, 2022-08-25:
@mriscoc 
the reason is because I saw [this video about it](https://www.youtube.com/watch?v=aMMYVVKzvO8)
[image]

mriscoc, 2022-08-25:
> @mriscoc the reason is because I saw [this video about it](https://www.youtube.com/watch?v=aMMYVVKzvO8)

The only case when you need to put the tramming point exactly over the screws is when you know the exactly screw thread, then enabling ASSISTED_TRAMMING can give you how many degrees or minutes you must turn your knob, but in the reality turn 7° 15" don't says too much but just a small turn.

classicrocker883, 2022-08-26:
thats good to know, i always wondered what that value was for if u have an M3 or M4 screw, thanks

closed 2022-08-26 by classicrocker883