mriscoc/Ender3V2S1 #307: Front left probing fails at tramming

stijnb1234, 2022-08-23:
**Describe the bug**
When I try tramming my bed using the CRTouch, the front left probing fails. The nozzle crashes into the bed before the probing stops. For all the other points, it works fine. I get the "Probe out of bounds" message.

**To Reproduce**
1. Power on the printer
2. Set the bed temperature to 60 degrees Celsius
3. Start the (automatic) bed tramming
4. See described bug for the front left position

**Expected behavior**
It should just probe without an error.

**Screenshots**
https://user-images.githubusercontent.com/27051051/186103300-21164dc7-0e51-41f6-bf3c-98987377f75f.mp4

**Version (please complete the following information):**
20220816 (August 17, 2022)

mriscoc, 2022-08-23:
Please read the Wiki: https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides

stijnb1234, 2022-08-23:
Is that something I can tweak in the settings or do I have to change my mount? I already followed the provided instructions.

ZipperZ, 2022-08-23:
Most likely the probe has to be brought down in reference with nozzle.

mriscoc, 2022-08-23:
[image]

stijnb1234, 2022-08-24:
Thanks for your help! I found out that my probe was a little loose in the mount. I've added threaded inserts and that fixed my problem.

closed 2022-08-24 by stijnb1234